alf45tar/PedalinoMini #75: OTA Not working

jimhiggs, 2020-02-21:
Env esp32doit-devkit-v1-wifi
Update from the web page not working. Monitor log: 

Update Start: firmware.bin
Update start
Progress:   0.0%
Progress:   0.0%
abort() was called at PC 0x4018165d on core 1

Backtrace: 0x4008d300:0x3ffd8f10 0x4008d531:0x3ffd8f30 0x4018165d:0x3ffd8f50 0x40087b03:0x3ffd8f70 0x40106701:0x3ffd8f90 0x401043fe:0x3ffd8fb0 0x401045dd:0x3ffd8fd0 0x400e8d16:0x3ffd8ff0 0x400d3735:0x3ffd9050 0x401b6eeb:0x3ffd9090 0x401b6bda:0x3ffd90e0 0x400f5481:0x3ffd9110 0x400f6ece:0x3ffd9180 0x400f70a5:0x3ffd91c0 0x401a3e91:0x3ffd91e0 0x401a3f0d:0x3ffd9220 0x401a460e:0x3ffd9240 0x40089889:0x3ffd9270

alf45tar, 2020-02-21:
OTA will not anymore supported because the size of the app is currently more than 2M.
The enable OTA we need at least 2 partitions that can containts the app but the total flash is only 4Mb and we need space for nvs and spiffs.

jimhiggs, 2020-02-21:
Is that true even if do a minimum build? I do not need
 iOS or Android app
Bluetooth
Network MIDI (aka AppleMIDI or RTP-MIDI)
ipMIDI
Open Sound Control (OSC)

alf45tar, 2020-02-22:
If you build the wifi (or ble) only version it is enough to reduce the size under 2M.

closed 2020-02-22 by alf45tar

jimhiggs, 2020-02-22:
I have been using the esp32doit-devkit-v1-wifi and am still getting the error.

alf45tar, 2020-02-22:
did you used the last version? I made a change to platformio.ini.

jimhiggs, 2020-02-22:
I think I did but will try again.

jimhiggs, 2020-02-22:
Using latest. It now seems to work in that in monitor it shows the progress going up  like so:
Progress:  69.0%
Progress:  69.0%
Progress:  69.2%
Progress:  69.2%
Progress:  69.4%
Progress:  69.4%
Progress:  69.4%
Progress:  69.6%
Progress:  69.6%
Progress:  69.6%
Progress:  69.8%
Progress:  69.8%
Update Success: 1373920B
Strange only goes up to 69.8%
But the oled screen just displays a mess when started and then goes black halfway through and needs a power reset to get it working again (board reset does not work)

jimhiggs, 2020-02-22:
Update. Tried again with power supply instead of usb power. The screen appears to be showing the update progress but it is messed up but it did not go black or need a power up this time.
So basically just something wrong with displaying on oled.