alf45tar/PedalinoMini #77: wifi smart config not work, can I add the wifi SSID/pwd in source code to compile a ready to use rom?

tonywk, 2020-02-24:
For some unknown reason, I can't use the wifi smart config method to complete the initial wifi setup, but the last AP method works, I can visit 192.168.4.1 web site, but the display format is different than normal look as the screen shot here (chrome browser)
1: If I can get help to resolve above issue
2: I wonder if I can add the wifi SSID/pwd in source code and compile the rom, after I burn the ESP32, it can work right away by connecting the pre-defined wifi

alf45tar, 2020-02-24:
1. There is no screenshot attached
2. Change the following lines and reset to factory default https://github.com/alf45tar/PedalinoMini/blob/523144304b89bbaaa7fd632f5608667d64de03da/src/Config.h#L28
https://github.com/alf45tar/PedalinoMini/blob/523144304b89bbaaa7fd632f5608667d64de03da/src/Config.h#L29

alf45tar, 2020-02-28:
It looks like you didn't Upload File System Image as in reported here

https://github.com/alf45tar/PedalinoMini#how-to-build-and-upload

Regarding E and F appearing on display is normal because you didn't connected a pull-up resistor to analog pins. You can avoid this changing pedal 5 and 6 to Momentary or to None.

Reboot happen when you select a mode with wifi and BLE because there is no enough memory to support both (for the moment).

It would be great if you can attach the log of PedalinoMini printed on serial (use Monitor in PlatformIO).

Thanks
alf45tar

alf45tar, 2020-02-29:
The latest version support hard coding SSID and password in Config.h.

tonywk, 2020-03-01:
Finally I get it working by erasing the esp32 and make a clean upload. I also changed another mobile phone to run esptouch and sucessfully configured the wifi. I just can't use normal boot mode as working state. At first time I complete the web configuration,after reboot to normal boot, it just reboot itself all the time. Using wifi or BLE mode is fine.  Is it due to low Free Heap?
Anyway, it is working in BLE only mode.

Copied below the monitor log.
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/
                                                                                  (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 7A286F24
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 259108, Internal Free Heap 230568
PSRAM Total Heap 0, PSRAM Free Heap 0
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:     7A286F24
[NVS][Global][Boot Mode]:       1
[NVS][Global][STA SSID]:        OpenWrt
[NVS][Global][STA Password]:    12345678
[NVS][Global][AP SSID]:         Pedalino-7A286F24
[NVS][Global][AP Password]:     7A286F24
[NVS][Global][Bootstrap Theme]: bootstrap
[NVS][Global][Current Profile]: 0
[NVS][Global][Tap Dance Mode]:  0
[NVS][Global][Bank Switch]:     0
[NVS][Global][Single Time]:     200
[NVS][Global][Double Time]:     400
[NVS][Global][Long   Time]:     500
[NVS][Global][Repeat Time]:     500
[NVS][Global][Blynk Cloud]:     0
[NVS][Global][Blynk Token]:

Boot NORMAL
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
Connecting to
SSID        : OpenWrt
Password    : 12345678
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
BSSID       : CC:81:DA:5C:0E:38
RSSI        : -39 dBm
Channel     : 10
STA         : 24:6F:28:7A:79:BC
SYSTEM_EVENT_STA_GOT_IP
Hostname    : 7A286F24
IP address  : 192.168.1.192
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.1.2
DNS 1       : 192.168.1.2
DNS 2       : 0.0.0.0
mDNS responder started
OTA update started
SPIFFS mount OK
HTTP server started
Connect to http://7A286F24.local/update for firmware update
Connect to http://7A286F24.local for configuration
ipMIDI server started
RTP-MIDI started
OSC server started
Internal Total Heap 251572, Internal Free Heap 36904
Loading profile ...
[... 104 lines cut ...]
Loading profile ...
Reading NVS Profile A ... done
Pedal autosensing...
Pedal  1   Tip Pin 25 Value 1023    Ring Pin 36 Value    0
Pedal  2   Tip Pin 26 Value 1023    Ring Pin 39 Value    0
Pedal  3   Tip Pin 27 Value 1023    Ring Pin 34 Value    0
Pedal  4   Tip Pin 14 Value 1023    Ring Pin 35 Value    0
Pedal  5   Tip Pin 12 Value 1023    Ring Pin 32 Value  939
Pedal  6   Tip Pin 13 Value 1023    Ring Pin 33 Value  905
Bank  1
Pedal  1     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      30   Channel  1   Pin D25
Pedal  2     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE       3   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      11   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1   Pin D14
Pedal  5     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      12   Channel  1   Pin D12
Pedal  6     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      13   Channel  1   Pin D13
MTC None
Pedal  5   input 1 output 1
Pedal  6   input 1 output 1
abort() was called at PC 0x4021887b on core 1

Backtrace: 0x40092e0c:0x3ffd1d90 0x4009303d:0x3ffd1db0 0x4021887b:0x3ffd1dd0 0x402188c2:0x3ffd1df0 0x402006e7:0x3ffd1e10 0x402007da:0x3ffd1e30 0x40200791:0x3ffd1e50 0x400f8cf7:0x3ffd1e70 0x400f8ff9:0x3ffd1e90 0x400f99e8:0x3ffd1ec0 0x400f99fd:0x3ffd1ee0 0x400eeb0e:0x3ffd1f00 0x40110421:0x3ffd1f20 0x4008f395:0x3ffd1f40

Rebooting...
ets Jun  8 2016 00:22:57

alf45tar, 2020-03-01:
Internal Free Heap 36904

Yes, it is. 36K are not enough to run Pedalino.

closed 2020-03-01 by alf45tar