alf45tar/PedalinoMini #78: Thru and Routing

athlonwai, 2020-02-24:
Hi! I really love your project, thanks for the hard work!

I wonder if you could explain a little more about the function thru and routing? If I turn on midi thru of, say, DIN usb, does it mean the midi message coming from DIN midi in will be forwarded to DIN midi out or does it also forward to USB Midi, ipmidi, etc?
How about routing? Am I able to route midi message from any inputs/outputs?
Thank you so much!

alf45tar, 2020-02-24:
Midi Routing has no effect for the moment because routing is on by default.
Every message received from any interface (with MIDI IN enabled) will be routed to all the other interfaces (with MIDI OUT enabled).

If MIDI THRU is enabled evey messages received from MIDI IN will be forwarder to MIDI OUT of the same interface.

athlonwai, 2020-02-26:
Thanks for your information!

I have got another problem: Smartconfig successfully connects but the SSID and password are not saving, everytime I reboot the device, it goes through the initial setup process looking for wifi and then smartconfig. The changes made using the webUI are not saving as well.
Here's what I see on the serial monitor:

Updating NVS Profile A[E][Preferences.cpp:272] putBytes(): nvs_set_blob fail: Banks NOT_ENOUGH_SPACE
[E][Preferences.cpp:272] putBytes(): nvs_set_blob fail: Pedals NOT_ENOUGH_SPACE
[E][Preferences.cpp:272] putBytes(): nvs_set_blob fail: Interfaces NOT_ENOUGH_SPACE
[E][Preferences.cpp:272] putBytes(): nvs_set_blob fail: Sequences NOT_ENOUGH_SPACE
[E][Preferences.cpp:112] putUChar(): nvs_set_u8 fail: Current Bank NOT_ENOUGH_SPACE
[E][Preferences.cpp:112] putUChar(): nvs_set_u8 fail: Current MTC NOT_ENOUGH_SPACE
[E][Preferences.cpp:112] putUChar(): nvs_set_u8 fail: Current Profile NOT_ENOUGH_SPACE

Do know what could be wrong? Thank you again!

alf45tar, 2020-02-26:
I would suggest to Erase Flash, Upload and Upload File System Image

athlonwai, 2020-02-26:
I tried this code, but i still cannot initiate nvs_flash :(

`void setup()
{
   Serial.begin(115200);

    ESP_ERROR_CHECK(nvs_flash_erase());
    while (!nvs_flash_init()){;
      Serial.println("Something is wrong");
      delay(500);
    }
    
}`

athlonwai, 2020-02-26:
Found [this example](https://github.com/espressif/arduino-esp32/tree/master/libraries/SPIFFS/examples/SPIFFS_Test) which test and format the flash memory, reupload Pedalinomini and it seems working now! Thanks again!

Btw, there's option to Upload File System Image? I remember it used to be there

alf45tar, 2020-02-26:
If you don't run “Upload File System image” task in PlatformIO IDE or use PlatformIO Core (CLI) and 'platformio run --target uploadfs' command you cannot use the web interface.

athlonwai, 2020-02-26:
Sorry for having so many questions,

```
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[W][AsyncTCP.cpp:950] _poll(): rx timeout 4
```

Are these errors? Everything else seems working probably but some random reboot occur:

```
abort() was called at PC 0x4020138b on core 1

Backtrace: 0x40092e0c:0x3ffd1c40 0x4009303d:0x3ffd1c60 0x4020138b:0x3ffd1c80 0x402013d2:0x3ffd1ca0 0x4020147f:0x3ffd1cc0 0x40201502:0x3ffd1ce0 0x40201519:0x3ffd1d00 0x400f9a37:0x3ffd1d20 0x400f9d39:0x3ffd1d40 0x400fa728:0x3ffd1d70 0x400fa73d:0x3ffd1d90 0x400ef806:0x3ffd1db0 0x401121f5:0x3ffd1dd0 0x4008f395:0x3ffd1df0

Rebooting...
```

alf45tar, 2020-02-28:
No, they aren't. They are warning and they are coming from AsyncTCP.cpp library. Search the internet for more details.

closed 2020-02-29 by alf45tar