alf45tar/PedalinoMini #79: OLED display faulty on update.

jimhiggs, 2020-02-27:
Latest version.
Display on update looks like this:

[image]

alf45tar, 2020-02-27:
Did you try with previous version?

jimhiggs, 2020-02-27:
Sorry yes, it has been like that for a while but before that I was crashing before writing to oled.

alf45tar, 2020-03-03:
Did you solved? If the issue is still presen check the hardware connections.

jimhiggs, 2020-03-03:
Unless you have a new update (less than 5 days old) then it is still present. I don't see how it can be a hardware issue as the oled works perfectly normally, it is only the update that screws up the screen.

alf45tar, 2020-03-04:
Now it is clear to me. The screen is corrupted during uploading the new firmware not after the update.
I don't have the issue. Are you using OTA update or via USB?

jimhiggs, 2020-03-04:
Sorry I was not clear. During OTA, the progress bar is corrupted.

alf45tar, 2020-03-10:
Display cannot be updated during firmware update via HTTP because there are no enough CPU resources. Update progress is not anymore reported on OLED but only on leds.
If you give a try to the latest version it should be fixed.

closed 2020-03-10 by alf45tar