alf45tar/PedalinoMini #80: How to change the Bank within the profile? E.g. with a button press. Web change Bank also cause reboot

tonywk, 2020-03-01:
I want to change the Bank with a button press, how to achieve it？

Another issue: I boot to wifi only mode, and can change the Bank to 2 in live tab. oled shows correct A.02, but with several button press, esp32 reboots and return back to A.01. From then on, it is not stable, reboots with several button press for many cycles. In BLE mode seems stable enough, with many button press, no reboots.

Log:
Internal Total Heap 254728, Internal Free Heap 132248
Loading profile ...
Reading NVS Profile A ... done
Pedal autosensing...
Pedal  1   Tip Pin 25 Value 1023    Ring Pin 36 Value    0
Pedal  2   Tip Pin 26 Value 1023    Ring Pin 39 Value    0
Pedal  3   Tip Pin 27 Value 1023    Ring Pin 34 Value    0
Pedal  4   Tip Pin 14 Value 1023    Ring Pin 35 Value    0
Pedal  5   Tip Pin 12 Value 1023    Ring Pin 32 Value  939
Pedal  6   Tip Pin 13 Value 1023    Ring Pin 33 Value  905
Bank  1
Pedal  1     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      30   Channel  1   Pin D25
Pedal  2     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE       3   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      11   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1   Pin D14
Pedal  5     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      12   Channel  1   Pin D12
Pedal  6     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      13   Channel  1   Pin D13
MTC None
Pedal  5   input 1 output 1
Pedal  6   input 1 output 1
/live 11641 bytes
HTML page lenght: 11641
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[W][AsyncTCP.cpp:930] _poll(): pcb is NULL
ws[/ws][1] connect
ws[/ws][1] binary-message[6]: 62 61 6e 6b 33 00
Pedal  3   input 0 output 0
NOTE ON.....Note  64.....Velocity 100.....Channel  2
Received from 192.168.1.192  NoteOn  0x40   Velocity 0x64   Channel 02
Pedal  3   input 1 output 1
NOTE OFF....Note  64......Velocity   0.....Channel  2
Received from 192.168.1.192  NoteOff 0x40   Velocity 0x00   Channel 02
Pedal  2   input 0 output 0
NOTE ON.....Note  62.....Velocity 100.....Channel  2
Received from 192.168.1.192  NoteOn  0x3E   Velocity 0x64   Channel 02
Pedal  2   input 1 output 1
NOTE OFF....Note  62......Velocity   0.....Channel  2
Received from 192.168.1.192  NoteOff 0x3E   Velocity 0x00   Channel 02
Pedal  1   input 0 output 0
NOTE ON.....Note  60.....Velocity 100.....Channel  2
Received from 192.168.1.192  NoteOn  0x3C   Velocity 0x64   Channel 02
Pedal  1   input 1 output 1
NOTE OFF....Note  60......Velocity   0.....Channel  2
Received from 192.168.1.192  NoteOff 0x3C   Velocity 0x00   Channel 02
[W][AsyncTCP.cpp:950] _poll(): rx timeout 4
Pedal  2   input 0 output 0
NOTE ON.....Note  62.....Velocity 100.....Channel  2
Received from 192.168.1.192  NoteOn  0x3E   Velocity 0x64   Channel 02
Pedal  2   input 1 output 1
NOTE OFF....Note  62......Velocity   0.....Channel  2
Received from 192.168.1.192  NoteOff 0x3E   Velocity 0x00   Channel 02
abort() was called at PC 0x4021887b on core 1

Backtrace: 0x40092e0c:0x3ffd1c60 0x4009303d:0x3ffd1c80 0x4021887b:0x3ffd1ca0 0x402188c2:0x3ffd1cc0 0x402006e7:0x3ffd1ce0 0x402007da:0x3ffd1d00 0x40200791:0x3ffd1d20 0x400f8cf7:0x3ffd1d40 0x400f8ff9:0x3ffd1d60 0x400f99e8:0x3ffd1d90 0x400f99fd:0x3ffd1db0 0x400eeb0e:0x3ffd1dd0 0x40110421:0x3ffd1df0 0x4008f395:0x3ffd1e10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 243740, Internal Free Heap 215316
PSRAM Total Heap 0, PSRAM Free Heap 0
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/
                                                                                  (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 7A286F24
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 259108, Internal Free Heap 230568
PSRAM Total Heap 0, PSRAM Free Heap 0
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:     7A286F24
[NVS][Global][Boot Mode]:       3
[NVS][Global][STA SSID]:        OpenWrt
[NVS][Global][STA Password]:    12345678
[NVS][Global][AP SSID]:         Pedalino-7A286F24
[NVS][Global][AP Password]:     7A286F24
[NVS][Global][Bootstrap Theme]: bootstrap
[NVS][Global][Current Profile]: 0
[NVS][Global][Tap Dance Mode]:  0
[NVS][Global][Bank Switch]:     0
[NVS][Global][Single Time]:     200
[NVS][Global][Double Time]:     400
[NVS][Global][Long   Time]:     500
[NVS][Global][Repeat Time]:     500
[NVS][Global][Blynk Cloud]:     0
[NVS][Global][Blynk Token]:

Boot WIFI
USB MIDI started
DIN MIDI started
Connecting to
SSID        : OpenWrt
Password    : 12345678
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
BSSID       : CC:81:DA:5C:0E:38
RSSI        : -31 dBm
Channel     : 10
STA         : 24:6F:28:7A:79:BC
SYSTEM_EVENT_STA_GOT_IP
Hostname    : 7A286F24
IP address  : 192.168.1.192
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.1.2
DNS 1       : 192.168.1.2
DNS 2       : 0.0.0.0
mDNS responder started
OTA update started
SPIFFS mount OK
HTTP server started
Connect to http://7A286F24.local/update for firmware update
Connect to http://7A286F24.local for configuration
ipMIDI server started
RTP-MIDI started
OSC server started
Internal Total Heap 254744, Internal Free Heap 132248
Loading profile ...
Reading NVS Profile A ... done
Pedal autosensing...
Pedal  1   Tip Pin 25 Value 1023    Ring Pin 36 Value    0
Pedal  2   Tip Pin 26 Value 1023    Ring Pin 39 Value    0
Pedal  3   Tip Pin 27 Value 1023    Ring Pin 34 Value    0
Pedal  4   Tip Pin 14 Value 1023    Ring Pin 35 Value    0
Pedal  5   Tip Pin 12 Value 1023    Ring Pin 32 Value  940
Pedal  6   Tip Pin 13 Value 1023    Ring Pin 33 Value  880
Bank  1
Pedal  1     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      30   Channel  1   Pin D25
Pedal  2     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE       3   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      11   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1   Pin D14
Pedal  5     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      12   Channel  1   Pin D12
Pedal  6     MIDI         MOMENTARY1   PRESS_1       POLARITY-   CONTROL_CHANGE      13   Channel  1   Pin D13
MTC None
Pedal  5   input 1 output 1
Pedal  6   input 1 output 1

alf45tar, 2020-03-01:
Live tab is under development and the reboot is a well know bug. I suggest to not use it.
Configure a pedal with Bank+1 pedal function in order to bank change.

tonywk, 2020-03-02:
Understood. Thanks for the info.

closed 2020-03-02 by tonywk