alf45tar/PedalinoMini #81: Installation difficulties

wrpooley, 2020-03-01:
Might not be the right place for this but I'm stuck early in the installation process.  I've installed Platformio into Visual Studio Code (Mac) but in trying to run the Git: Clone command I get an error the command is not found.  It seems no commands are found either.  I am more familiar with the Arduino IDE so forgive my ignorance!  Looking forward to tinkering with this project.

Intended hardware is an Espressif WROOM32D or an ADAFruit Huzzah32.

alf45tar, 2020-03-01:
Did you installed git from https://git-scm.com/?

wrpooley, 2020-03-01:
No. I missed that step :).  doing it now...

wrpooley, 2020-03-01:
Still seeing 'command not found'.  Do I need to configure Git?

wrpooley, 2020-03-03:
Still love to get some guidance on this.  If someone could walk me through the installation procedure with a bit more detail I'd greatly appreciate it

alf45tar, 2020-03-04:
If Git:Clone doesn't work I can suggest to download the zip file from github, unzip into a folder and open the folder with VSC.

wrpooley, 2020-03-04:
I've got it installed (sort of).  I hadn't yet installed the Command Line editor on my Mac.  However...

After many attempts to upload I've yet to get BLE active.  In some cases the serial monitor shows the service is advertising but it does not show up as a device on my phone, etc.  

I tried commenting out different boards during the upload and and there are often 'problems' but I still get a successful upload - or so it reports.

I've tried with 2 boards: ESP32 DEVKIT 32D (http://hiletgo.com/ProductDetail/1906566.html) and the adafruit Huzzah32 (https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview).

Also, even though I can ultimately open the web page configuration file, I've not been successful in connecting it either to my laptop (Mac 10.14.6) or my iPhoneXS.

Any ideas?

alf45tar, 2020-03-04:
Which env did you uploaded? Attach here your platformio.ini and the serial log to receive support.

wrpooley, 2020-03-04:
I upload both the esp32doit-devkit-v1-ble and the esp32doit-devkit-v1-wifi.  
[platformio.ini.zip](https://github.com/alf45tar/PedalinoMini/files/4286611/platformio.ini.zip)

This is the serial log after reboot (I'll have to try uploading again to reproduce that serial log):

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 243872, Internal Free Heap 215348
PSRAM Total Heap 0, PSRAM Free Heap 0
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: B6286F24
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 259240, Internal Free Heap 230600
PSRAM Total Heap 0, PSRAM Free Heap 0
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:     B6286F24
[NVS][Global][Boot Mode]:       3
[NVS][Global][STA SSID]:        
[NVS][Global][STA Password]:    
[NVS][Global][AP SSID]:         Pedalino-B6286F24
[NVS][Global][AP Password]:     B6286F24
[NVS][Global][Bootstrap Theme]: bootstrap
[NVS][Global][Current Profile]: 0
[NVS][Global][Tap Dance Mode]:  0
[NVS][Global][Bank Switch]:     0
[NVS][Global][Single Time]:     200
[NVS][Global][Double Time]:     400
[NVS][Global][Long   Time]:     500
[NVS][Global][Repeat Time]:     500
[NVS][Global][Blynk Cloud]:     0
[NVS][Global][Blynk Token]:                                     
[NVS][Global][Ladder]:          Ladder 0 Threshold 497 Tolerance 82
[NVS][Global][Ladder]:          Ladder 1 Threshold 660 Tolerance 47
[NVS][Global][Ladder]:          Ladder 2 Threshold 752 Tolerance 33
[NVS][Global][Ladder]:          Ladder 3 Threshold 816 Tolerance 31
[NVS][Global][Ladder]:          Ladder 4 Threshold 876 Tolerance 31
[NVS][Global][Ladder]:          Ladder 5 Threshold 945 Tolerance 35

Boot WIFI
USB MIDI started
DIN MIDI started
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SmartConfig started
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
SmartConfig timeout
WPS started
WPS timeout
AP Pedalino-B6286F24 started with password B6286F24
AP SSID     : Pedalino-B6286F24
AP PSK      : B6286F24
SYSTEM_EVENT_AP_START
AP MAC      : 24:6F:28:B6:19:C9
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
AP IP       : 192.168.4.1
Channel     : 1
Connect to Pedalino-B6286F24 wireless network with password B6286F24
mDNS responder started
OTA update started
SPIFFS mount OK
HTTP server started
Connect to http://B6286F24.local/update for firmware update
Connect to http://B6286F24.local for configuration
ipMIDI server started
RTP-MIDI started
OSC server started
Internal Total Heap 254408, Internal Free Heap 115296
Loading profile ...
Reading NVS Profile A ... done
Pedal autosensing...
Pedal  1   autosensing disabled
Pedal  2   autosensing disabled
Pedal  3   autosensing disabled
Pedal  4   autosensing disabled
Pedal  5   autosensing disabled
Pedal  6   autosensing disabled
Bank  1
Pedal  1     MIDI                      PRESS_1       POLARITY+   NOTE_ON_OFF         60   Channel  1
Pedal  2     MIDI                      PRESS_1       POLARITY+   NOTE_ON_OFF         62   Channel  1
Pedal  3     MIDI                      PRESS_1       POLARITY+   NOTE_ON_OFF         64   Channel  1
Pedal  4     MIDI                      PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1
Pedal  5     MIDI                      PRESS_1       POLARITY-   CONTROL_CHANGE      12   Channel  1
Pedal  6     MIDI                      PRESS_1       POLARITY-   CONTROL_CHANGE      13   Channel  1
MTC None
Received from Serial MIDI  AfterTouchChannel   Pressure 0x80   Channel 07
Received from Serial MIDI  AfterTouchChannel   Pressure 0x00   Channel 07
Received from Serial MIDI  AfterTouchChannel   Pressure 0x40   Channel 07
Received from Serial MIDI  AfterTouchChannel   Pressure 0x00   Channel 07
Received from Serial MIDI  NoteOff 0x00   Velocity 0x00   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x00   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x00   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x00   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x02   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x00   Channel 01
Received from Serial MIDI  NoteOff 0x00   Velocity 0x40   Channel 01

wrpooley, 2020-03-04:
this is the serial log after Build:

Can not remove temporary directory `/Users/waynep/Documents/Arduino/PedalinoMini-1/.pio/build`. Please remove it manually to avoid build issues
Processing esp32doit-devkit-v1-ble (board: esp32doit-devkit-v1; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.2 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AppleMIDI> 1.6.1
|-- <Battery Sense> 1.1.2
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Bounce2> 2.53
|-- <AsyncTCP> 1.1.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ESP8266_SSD1306> 4.1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <MD_UISwitch> 2.1.0
|-- <MD_REncoder> 1.0.1
|-- <MIDI Library> 4.3.1 #4daf363
|-- <OSC>
|-- <RemoteDebug> 3.0.5
|   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ResponsiveAnalogRead> 1.2.1
|-- <BLE-MIDI> 0.0.1 #6eaf748
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <ESP32 Async UDP> 1.0.0
|-- <Update> 1.0
|-- <Ticker> 1.1
|-- <FS> 1.0
|-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
Building in release mode
Retrieving maximum program size .pio/build/esp32doit-devkit-v1-ble/firmware.elf
Checking size .pio/build/esp32doit-devkit-v1-ble/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.3% (used 40368 bytes from 327680 bytes)
Flash: [=====     ]  54.8% (used 1078359 bytes from 1966080 bytes)
============================================================================= [SUCCESS] Took 5.26 seconds =============================================================================

Processing esp32doit-devkit-v1-wifi (board: esp32doit-devkit-v1; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.2 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AppleMIDI> 1.6.1
|-- <Battery Sense> 1.1.2
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Bounce2> 2.53
|-- <AsyncTCP> 1.1.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ESP8266_SSD1306> 4.1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <MD_UISwitch> 2.1.0
|-- <MD_REncoder> 1.0.1
|-- <MIDI Library> 4.3.1 #4daf363
|-- <OSC>
|-- <RemoteDebug> 3.0.5
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ResponsiveAnalogRead> 1.2.1
|-- <BLE-MIDI> 0.0.1 #6eaf748
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESP32 Async UDP> 1.0.0
|-- <Preferences> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Ticker> 1.1
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
Building in release mode
Linking .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Building .pio/build/esp32doit-devkit-v1-wifi/firmware.bin
Retrieving maximum program size .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Checking size .pio/build/esp32doit-devkit-v1-wifi/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.8% (used 55080 bytes from 327680 bytes)
Flash: [======    ]  61.7% (used 1212717 bytes from 1966080 bytes)
esptool.py v2.6
============================================================================= [SUCCESS] Took 10.94 seconds =============================================================================

Environment               Status    Duration
------------------------  --------  ------------
esp32dev                  IGNORED
esp32doit-devkit-v1       IGNORED
esp32doit-devkit-v1-ble   SUCCESS   00:00:05.263
esp32doit-devkit-v1-wifi  SUCCESS   00:00:10.943
ttgo-t-eight              IGNORED
ttgo-t-eight-ble          IGNORED
ttgo-t-eight-wifi         IGNORED
============================================================================= 2 succeeded in 00:00:16.206 =============================================================================

alf45tar, 2020-03-04:
First of all select the version you want to test and upload it.
If you upload the ble version wifi and web config will not available.

However the attached log is for the wifi version in AP mode (without BLE of course). Didi you connected to the Pedalino hotspot as indicated in the log?

> Connect to Pedalino-B6286F24 wireless network with password B6286F24

In any case you built the wifi version with NOWEBCONFIG flag. The web UI will not available.

wrpooley, 2020-03-04:
Hmm.  I see what you mean about NOWEBCONFIG flag.  I tried to upload both the ble and the wifi versions and this is the serial log from that.  Also, I set it to Normal Boot mode.  Even though it reports [SUCCESS] I think there may be something wrong this but I'm not nearly experienced enough to figure it out!  :)

> Executing task: platformio run --target upload --target monitor <

Can not remove temporary directory `/Users/waynep/Documents/Arduino/PedalinoMini-1/.pio/build`. Please remove it manually to avoid build issues
Processing esp32doit-devkit-v1-ble (board: esp32doit-devkit-v1; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.2 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AppleMIDI> 1.6.1
|-- <Battery Sense> 1.1.2
|-- <Blynk> 0.6.1
|   |-- <WiFi> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Bounce2> 2.53
|-- <AsyncTCP> 1.1.1
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <ESP8266_SSD1306> 4.1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <MD_UISwitch> 2.1.0
|-- <MD_REncoder> 1.0.1
|-- <MIDI Library> 4.3.1 #4daf363
|-- <OSC>
|-- <RemoteDebug> 3.0.5
|   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <ResponsiveAnalogRead> 1.2.1
|-- <BLE-MIDI> 0.0.1 #6eaf748
|   |-- <ESP32 BLE Arduino> 1.0.1
|-- <ESP32 Async UDP> 1.0.0
|-- <Update> 1.0
|-- <FS> 1.0
|-- <Ticker> 1.1
|-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
Building in release mode
Compiling .pio/build/esp32doit-devkit-v1-ble/src/MidiTimeCode.cpp.o
Compiling .pio/build/esp32doit-devkit-v1-ble/src/PedalinoMini.cpp.o
Compiling .pio/build/esp32doit-devkit-v1-ble/lib4cd/AppleMIDI_ID222/AppleMidi.cpp.o
Compiling .pio/build/esp32doit-devkit-v1-ble/lib947/Battery Sense_ID735/Battery.cpp.o
[... 513 lines cut ...]
Pedal  5   input 48 output 127 velocity 5.02
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Loading profile ...
Reading NVS Profile B ... done
Pedal autosensing...
Pedal  1   autosensing disabled
Pedal  2   autosensing disabled
Pedal  3   autosensing disabled
Pedal  4   autosensing disabled
Pedal  5   Tip Pin 12 Value 1023    Ring Pin 32 Value  754
Pedal  6   Tip Pin 13 Value 1023    Ring Pin 33 Value  628
Bank  1
Pedal  1     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         60   Channel  1   Pin D25
Pedal  2     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         62   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         64   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1   Pin D14
Pedal  5     MIDI         ANALOG       PRESS_1       POLARITY-   CONTROL_CHANGE      12   Channel  1   Pin A32 D12
Pedal  6     MIDI         ANALOG       PRESS_1       POLARITY-   CONTROL_CHANGE      13   Channel  1   Pin A33 D13
Pedal  5 calibration min 83
Pedal  5 calibration max 83
Pedal  6 calibration min 0
Pedal  5 calibration min 13
Pedal  5 calibration min 0
MTC None

alf45tar, 2020-03-04:
It looks you are running the BLE version (esp32doit-devkit-v1-ble).
Web configuration is not available but PedaliniMini should be discoverable by your iPhone. You can try with Garageband for example.

wrpooley, 2020-03-05:
I’m admittedly fumbling in the dark a bit!  After your most recent reply I looked for a BLE device from within a music app like you suggested and to my surprise I did see the pedaling! I incorrectly was looking for a device in my Bluetooth pairing. It was VERY late (EST) so I didn’t confirm it it was working and it didn’t stay connected long. I went to bed after that. 

However I can’t seem to be able to get back to that place where I see it being advertised.  Can you clarify a few things?

Should I be trying to install multiple ‘env’s? I’ve confirmed the whenever I include the TTGO env the board just reboots over and over again. 

I’ve tried commenting everything but the ESP32-ble and I’ve never been able to see that service advertise again. 

Is the concept to upload multiple ‘builds’ and then choose which one is active at the boot up?  

Apologies for all the questions but this is pretty much all over my head but I am learning 😀

mknerr, 2020-03-05:
I would comment out the builds you don’t need, if for no other reason, than to prevent having to wait for them to build each time. However, PlatformIO will only upload the ones that match your board. Then, yes, you can select Bluetooth, WiFi, or Both on startup, with the caveat that the web interface will make PedalinoMini crash if BT and WiFi are loaded at the same time.

wrpooley, 2020-03-05:
Thanks for the response!  Any recommendations for troubleshooting the lack of BLE?  I felt like there was somewhere in the serial monitor that showed when the service was activated but I can’t see that find that anymore. Come to think of it I also haven’t successfully sent any ipMidi either as my Mac throws up an error saying no midi was received/check firewall - I’ve seen this under other circumstance so it may be a Mojave thing. But my phone doesn’t seem to connect either.

alf45tar, 2020-03-05:
First of all try the latest version just uploaded and reset Pedalino to factory default.

You don't need to build everything and upload everything. You can select which version you want to build and upload.
[image]

If it is still not working attach the log (only the boot log not the compile log).

Thanks
alf45tar

wrpooley, 2020-03-06:
Thanks for the suggestions on how to focus on specific builds.  I seem to be able to make BLE connections from Pedalino to my phone but no evidence it's going the other way.  Is there a way to confirm that it's receiving anything?  One thing to add is that even though I've set all but one pedal to none and turned off auto-sensing everywhere, I end up seeing a lot of spurious data being sent over BLE.  NOTE: I have only wired a single switch to D1 on a breadboard.  Any thoughts on where/how all the extra midi data is being generated?

alf45tar, 2020-03-06:
Try the following app to see incoming midi messages
https://apps.apple.com/us/app/midi-wrench/id589243566

or AmpliTube to control it
https://apps.apple.com/it/app/amplitube-cs/id373309342

wrpooley, 2020-03-06:
I am familiar with those apps and actually use midi wrench. I use an app called midi pipe to inspect incoming messages on my MacBook Pro. Band helper for iOS has a pretty deep midi implementation so I can inspect there as well. 

Interestingly, it looks like all of the spurious midi noise is coming displayed as ‘Received from Serial MIDI’. I’ve removed all switches in each profile except for one in A. This ‘noise’ only shows up when I connect to either my Mac or phone over BLE. 

Is the PedalinoMini display (or serial monitor) supposed to display incoming midi messages?  It does not for me at least. 

No luck on network midi either. OSC or RTP. 

I tried building from scratch with the latest and no change in behavior.

alf45tar, 2020-03-07:
If you didn't implemented the Serial MIDI interface disable from web config to avoid spurious incoming messages.

>Is the PedalinoMini display (or serial monitor) supposed to display incoming midi messages?

Yes, only for ipMidi for the moment, Is this feature important for you?

alf45tar, 2020-03-07:
Added display incoming MIDI messages for Serial, USB and RTC.

wrpooley, 2020-03-07:
How does one disable the serial midi hardware interface?  

Having the display show incoming messages was more a way to troubleshoot BLE and Network Midi connections. So far I’ve had limited success making BLE connections but no Network/OSC as of yet.  OSC is of more interest to me at the moment.

alf45tar, 2020-03-10:
Boot in WiFi mode and use the web ui.

wrpooley, 2020-03-10:
I'm making progress at least!  I figured out that I'm not able to make network midi (& OSC) connections when in AP mode however I can if I join an existing network.  

My next hurdle is to eradicate all the Serial MIDI data that makes makes the screen go crazy!  I can't seem to turn it off and I don't see a pattern in it.  Attached is a sample of what the serial monitor is seeing.

[Untitled.txt](https://github.com/alf45tar/PedalinoMini/files/4315062/Untitled.txt)

UPDATE:  I commented this part from PedalinoMini.cpp and it did stop screen from showing extra data.  

  // Listen to incoming messages
/*

  if (USB_MIDI.read())
    DPRINTMIDI("USB MIDI", USB_MIDI.getType(), USB_MIDI.getChannel(), USB_MIDI.getData1(), USB_MIDI.getData2());
  if (DIN_MIDI.read())
    DPRINTMIDI("Serial MIDI", DIN_MIDI.getType(), DIN_MIDI.getChannel(), DIN_MIDI.getData1(), DIN_MIDI.getData2());

*/
  if (wifiEnabled) {
    // Listen to incoming AppleMIDI messages from WiFi
    rtpMIDI_listen();

One more thing:  How do you get the display to show what note/controller/Program change is being triggered in addition to the bank name (A, for example)?

wrpooley, 2020-03-10:
disregard question about display showing details.  I figured out removing the name field accomplished this.

closed 2020-03-11 by alf45tar