alf45tar/PedalinoMini #84: Problem Bluetooth mode and AP and Bluetooth mode

juliowebhouse, 2020-03-03:
Hello,

I'm using an ESP8266 arduino, I was able to upload the code, but when I put it in Bluetooth mode, the arduino keeps restarting and when I put it in AP and Bluetooth mode, neither my iPad nor my iPhone can locate the device. You can help me, thanks.

alf45tar, 2020-03-03:
Are you sure you are using an ESP8266? PedalinoMini works only on ESP32.
Supposing you are using an ESP32 board send me the error log when boot in BLE mode.

AP+BLE is not stable because there is not enough memory. The situation could change when IDF 4.0 with new BLE stack will be supported in Arduino framework.

juliowebhouse, 2020-03-04:
Hi, 

Sorry my board is a ESP32 DEV Kit1

This is LOG of the problem... Thanks

[Rebooting_bluetooth.txt](https://github.com/alf45tar/PedalinoMini/files/4287539/Rebooting_bluetooth.txt)

alf45tar, 2020-03-04:
Are you using the latest version of the firmware?

juliowebhouse, 2020-03-04:
Yes, I downloaded and installed it last week.

Thanks.

alf45tar, 2020-03-04:
Try the latest one. I release update daily.

juliowebhouse, 2020-03-05:
Hello,

Thanks for the help, I have no more problems with the restart, but neither the iPhone nor the iPad does not find the Pedalino via bluetooth. :(

This is new LOG... Thanks

[Rebooting_bluetooth.txt](https://github.com/alf45tar/PedalinoMini/files/4292483/Rebooting_bluetooth.txt)

alf45tar, 2020-03-05:
Try with Garageband using the following procedure:

https://support.apple.com/guide/garageband-ipad/a-bluetooth-midi-device-touch-instruments-chse356a0321/ipados

juliowebhouse, 2020-03-05:
alf45tar,

I try the step by step this link but not work, nor does the garageband find the BLE.

juliowebhouse, 2020-03-06:
Hello, good morning

I put a LOG in that part of the code and saw that it always returns FALSE.

#ifdef BLE
       bleEnabled = true;
#else
       bleEnabled = false;

Can you help me! Thanks

alf45tar, 2020-03-06:
Which version did you uploaded?

Have a look here https://github.com/alf45tar/PedalinoMini/issues/81#issuecomment-595100502

juliowebhouse, 2020-03-06:
Great thank you! The Pedalino now works with bluetooth!!!

closed 2020-03-07 by alf45tar