alf45tar/PedalinoMini #85: Problem to control Garageband and Amplitube on iPad

juliowebhouse, 2020-03-09:
Hello,

Now I can connect and use the pedal via the Macbook (I use the MainStage on the Macbook) using the BLE connection but on my iPad I can connect but I cannot configure the pedals, neither on it nor on GarageBand, on GarageBand I can see the Pedalino and connect me but I can't configure any pedal. Could you help me, thanks.

alf45tar, 2020-03-10:
You cannot configure the pedals with GarageBand on iPad. It is not possible to assign MIDI to pedals.
Try Amplitube. It works like a charm.

juliowebhouse, 2020-03-10:
Hello,

With Amplitube I can't connect via BLE but with Wifi and Midi Network yes, but the commands in the APP don't work.

:(

alf45tar, 2020-03-10:
Use GarageBand (or MIDI Wrench or midimttr) to connect via BLE and after switch to Amplitube.

https://apps.apple.com/us/app/midi-wrench/id589243566
https://apps.apple.com/it/app/midimittr/id925495245

juliowebhouse, 2020-03-10:
I followed these instructions and it doesn't work, I can connect via BLE using all 3 apps but within Amplitube I still can't assign any control.

juliowebhouse, 2020-03-10:
LOGS:

BLE MIDI Device Connected
Pedal  1   input 0 output 0
NOTE ON.....Note  60.....Velocity 100.....Channel  1
Pedal  1   input 1 output 1
NOTE OFF....Note  60......Velocity   0.....Channel  1
Pedal  2   input 0 output 0
NOTE ON.....Note  62.....Velocity 100.....Channel  1
Pedal  2   input 1 output 1
NOTE OFF....Note  62......Velocity   0.....Channel  1
Pedal  4   input 0 output 0
NOTE ON.....Note  65.....Velocity 100.....Channel  1
Pedal  4   input 1 output 1
NOTE OFF....Note  65......Velocity   0.....Channel  1

alf45tar, 2020-03-10:
I am not sure Amplitube can assign Note On/Off.
Did you try with Control Code?

juliowebhouse, 2020-03-10:
I finally made it! Works with control change. Thank you.

closed 2020-03-10 by alf45tar