alf45tar/PedalinoMini #87: Additional function when press 2 button once time

iwbrhwfy-source, 2020-03-17:
I hope new version you will enhance
I think it will save amount of push buttons.
Example button 1 and 2 for bank up, button 2 and 3 for bank down, button 1 and 2 and 3 for tuner :D
Thank you so much

iwbrhwfy-source, 2020-03-17:
and button seem not debounce when i select Bank+1, Bank-1  mode (sometime it from bank 1 to bank 3 or 4 with 1 click)
it still ok when i select MIDI mode

alf45tar, 2020-03-17:
Thanks for the suggestion.

Regarding your debounce issue please check your hardware and/or the software options. Whcih values are you using for Press Time, Double Press, Long Press and Repeat? Can you attach the the log?

Thanks
alf45tar

iwbrhwfy-source, 2020-03-17:
Thank for your reply
Hardware still ok because a changed some button, and when i select MIDI mode everything very good, only bank+ bank - have issue (i set min 1, max 10)
All value Press Time, Double Press, Long Press and Repeat is default
And how to export the log :D

iwbrhwfy-source, 2020-03-18:
https://youtu.be/wt8hBgezhzM
I don't extract log becase bank+ function not show on serial monitor @@

alf45tar, 2020-03-18:
Thank you for the video. Very useful.
You are using a latch switch. Change Pedal Mode from Momentary to Latch.

iwbrhwfy-source, 2020-03-18:
> Thank you for the video. Very useful.
> You are using a latch switch. Change Pedal Mode from Momentary to Latch.

No, it is momentary switch :D 
Midi function toggle between midi value 1 and midi value 2

alf45tar, 2020-03-18:
Can you try with Latch setting?

iwbrhwfy-source, 2020-03-18:
> Can you try with Latch 
https://youtu.be/VSq-OpDNgg8
Yes and i change to another button

alf45tar, 2020-03-18:
Can you check and confirm the following value into Options?

[image]

iwbrhwfy-source, 2020-03-18:
> Can you check and confirm the following value into Options?
> 
> [image]

I just erase flash and upload lastest firmware and still have debounce issue
[image]

iwbrhwfy-source, 2020-03-21:
> Can you check and confirm the following value into Options?
> 
> [image]

I still have issue  :((
https://youtu.be/4vpfVhldKb8

alf45tar, 2020-03-21:
Let's suppose that your switches are more noisy than normal.
I suggest to increase the debouncing interval from 20 ms to 50 ms or more.

https://github.com/alf45tar/PedalinoMini/blob/0138b4f3c949a73c0d5f26d99fc8bcab838f8e12/src/Pedalino.h#L105

Try to comment the following line
https://github.com/alf45tar/PedalinoMini/blob/0138b4f3c949a73c0d5f26d99fc8bcab838f8e12/src/Pedalino.h#L106

and uncomment the next one
https://github.com/alf45tar/PedalinoMini/blob/0138b4f3c949a73c0d5f26d99fc8bcab838f8e12/src/Pedalino.h#L107

iwbrhwfy-source, 2020-03-21:
> suppose

[image]
i still noise 
maybe i will try diffrent footswitch type

iwbrhwfy-source, 2020-03-21:
But it always work OK with any funtion except (Midi+1 2 3...)
https://youtu.be/7t95DVEmuN0

alf45tar, 2020-03-21:
Are you sure it works ok with MIDI function? 
Check the log if the MIDI message is sent only one time or 2 or more.

iwbrhwfy-source, 2020-03-22:
> Are you sure it works ok with MIDI function?
> Check the log if the MIDI message is sent only one time or 2 or more.

log with 3 button
[image]

alf45tar, 2020-03-24:
Did you modified my code?

iwbrhwfy-source, 2020-03-24:
> Did you modified my code?

No sir i clone new code everyday to test, and in 2 esp32 board different, i will test in another PC

iwbrhwfy-source, 2020-03-24:
> Did you modified my code?

I just change to another pc, install new VSC, clone new code and upload
It still error @@
https://youtu.be/7t95DVEmuN0

iwbrhwfy-source, 2020-03-24:
[image]
Reason from my button sir, it ok with small button
But i still down know why code can debounce with MIDI function but can't debounce BANK function

alf45tar, 2020-07-06:
I suggest to check the latest version 2.0 just released

closed 2020-07-06 by alf45tar