alf45tar/PedalinoMini #88: Noise when connected to esp32

patricksebastien, 2020-03-19:
Hi, I have so many questions... not sure if it's the right place to ask. Some of them are pretty basic. Anyway here we go with a first one.

I am using the wifi in access point, when connecting from my computer to Pedalino esp32, I receive a lots of Serial stuff on the log. Did a video (the blue led is when connected).

https://imgur.com/JfDjeeS

I am using the esp32s, here's the pinout:
[image]

Let me know if there's a forum or a better place to ask questions.
Thanks!

iwbrhwfy-source, 2020-03-20:
Sometime i have issue like you

patricksebastien, 2020-03-20:
EDIT:
Touching GIOP15 flood the log too, so I disabled the serial_midi:
_PedalinoMini.cpp_
Line 365 //serial_midi_connect();
&
Line 492 //if (*_MIDI.read())

mknerr, 2020-03-20:
https://github.com/alf45tar/PedalinoMini/issues/59#issue-547225208

patricksebastien, 2020-03-20:
@mknerr yep thanks for pointing this information, will need to retest with the pull-up resistor on all ADC, but in my case I already disabled all the pedals using the web-config. I think it was related to the midi serial (maybe my devboard is not 100% compatible with the esp32dev)...

iwbrhwfy-source, 2020-03-20:
In my case when i touch my hand to ground it will ok

alf45tar, 2020-03-20:
I did some changes to the source code.
When a serial MIDI interface (USB or Legacy) is disabled as input the received data are not logged or show on OLED.

By default MIDI IN is disabled for USB and Legacy interfaces because not all are implementing them.

closed 2020-03-21 by alf45tar