alf45tar/PedalinoMini #89: Lowest latency

patricksebastien, 2020-03-20:
Hi,
I want to build a looper using PedalinoMini (using Pure Data or SooperLooper). What is the lowest latency configuration that I can use, is it: Wifi or Bluetooth. If Wifi is it OSC (that would be great) or ipMIDI or RTP-Midi (not sure if there's a difference).

Anyone measure the latency? Is the distance playing a huge factor in the responsiveness?

alf45tar, 2020-03-25:
I don't know how to measure the latency in a precise way.

Using a Mac with Catalina and MIDI Network (AppleMIDI) protocol the typical latency reported by the software is between 5 and 10ms. Sometime is around 3 ms but some packet can be delayed until 50-100 ms.
Of course the overload of the WiFi and the distance from AP makes a big difference.

[image]

I think ipMidi could save some ms but it doesn't make any difference on high WiFi traffic load.
Bluetooth should be better but I don't know how to measure latency.

patricksebastien, 2020-03-25:
Thanks for the feedback. Will see if I can do a simple test (not very accurate but will give me an idea of what to expect): adc microphone close to a momentary switch -> pulse in speaker recorded with the same microphone (then in audacity I can calculated the latency).

A good read for bluetooth midi latency (lots of information + a test rig):
https://www.nime.org/proceedings/2019/nime2019_paper006.pdf

patricksebastien, 2020-03-27:
I had the time to dig this subject I have good news. I'm not 100% confident on the setup I did but here we go with the information I have.

Ubuntu Studio - low latency kernel running jack at 44100hz / 64 period / 2 buffer. Reported latency: 2.9ms, real latency (round-trip) reported by jack_iodelay:
```
457.089 frames     10.365 ms total roundtrip latency
	extra loopback latency: 265 frames
	use 132 for the backend arguments -I and -O
```

In PD I created a patch that send a pulse (sample accurate dirac impulse generator) when receiving a trigger, a microphone record the physical trigger on the device and both channels are sent to Audacity. Here's the patch for reference:

[image]

1) Commercial foot pedal midi (not USB) -> midi in of soundcard -> pd
[image]

2) Commercial USB finger pad trigger -> pd (via alsa midi)
[image]

3) PedalinoMini Wifi OSC
[image]

22ms very stable, the circuit was around 2 meters from my computer. I wasn't able to test farther (not enough microphone cable). I will not test BLE, because OSC is fast enough for my needs and very easy to receive in PD.

Cheers

MaxZak2020, 2021-01-04:
I have sad experience with looper in Ableton Live, wich I trigger with PedalinoMINI from MIDI out -> MIDI USB adapter. The latency is tremendows - about 1sec or more. It's imposible to use. How can it be and what to do? 
Thanks in advance for any ideas.

patricksebastien, 2021-10-30:
@he-leon I would like to learn more about your setup (sooperlooper + pedalino). Do you have a website where you share your work or a way to contact you?

he-leon, 2021-11-01:
> @he-leon I would like to learn more about your setup (sooperlooper + pedalino). Do you have a website where you share your work or a way to contact you?

I'm currently trying to find a good setup for sooperlooper in combination with PedalinoMini. The thing is I just started playing around with looping and I've never used a proper hardware looper  as a reference. So I'm not really sure how a good setup should feel. In differents words: If my loops don't seam well or are out of time it's hard to tell whether it's due to a bad software/hardware setup or because I just simply suck at looping, lol. 
Currently I'm using SooperLooper running on Arch Linux (no special low latency kernel). As an audio interface atm I use a cheap and crappy Behringer XENYX302USB. I built a small Pedalino board from some multiplex plywood with 5 momentary switches and an ESP32 devkit v1. I'm using the MIDI legacy DIN output of PedalinoMini connected to a Roland UM-One MkII USB MIDI interface. 
Currently I'm trying to figure out a way to get Input and Output latency compensation in Jack/SooperLooper dialed (I think theoretically you can compensate all MIDI and audio latency here. But not 100% sure.). 
I'd like to create a test setup for the loop station: As mentioned I might suck at looping which means I'm not a good reference for testing. That's why I'm thinking of programming an Arduino or ESP as  a simulated "looping artist" with perfect timing: It would push the switch (tie Pedalino input low) and play a click track for my audio interface simultaneously. By analyzing the recorded  audio in SooperLooper I could then measure any latency and compensate it accordingly. That's somewhat similar to your approach recording the "press sound" of the button and triggering a signal with Pure Data.

However, I'm not sure when I will find the time to do this. If I find a good setup eventually I'll update this thread or post a link to a write up! If you have any questions you can also contact me via email:  [image]

Cheers

closed 2026-11-02 by alf45tar