alf45tar/PedalinoMini #90: Separate program changes via double and long press

mknerr, 2020-03-20:
In the same way you can with CCs, is it possible to have separate program changes via double and long press? Don’t think so, but thought I’d ask. Thanks!

iwbrhwfy-source, 2020-03-20:
> In the same way you can with CCs, is it possible to have separate program changes via double and long press? Don’t think so, but thought I’d ask. Thanks!

you can edit some code, if longpress bank 1 it will send data in bank 5
[image]

mknerr, 2020-03-20:
Thank you, that could be an option. Would really like to keep the code stock, if possible, for now at least. 
I’d definitely like to get a handle on what is doing what at some point, though.

alf45tar, 2020-03-21:
Added with the lastest version

closed 2020-03-21 by alf45tar