alf45tar/PedalinoMini #92: SSH1106 OLED doesn't display anything

rarbuthnot, 2020-03-24:
Hi, thank you for this project. I'm just getting started, and I am very excited to see what is possible.

I have PedalinoMini running on a HiLetgo ESP-WROOM-32D Development Board, but nothing is showing on my HiLetgo 1.3" IIC I2C Serial 128x64 SSH1106 OLED.

I read that I needed to change the build flags in platformio.ini from SSD1306WIRE to SSH1106WIRE, so I did that, recompiled and updated, but am not seeing a change. I still see a completely black screen. I checked that the VCC pin is getting voltage and it is.

Please let me know if there are other changes I need to make. Thanks in advance!

alf45tar, 2020-03-24:
May I suggest to change the address from 0x3c to 0x78?

https://github.com/alf45tar/PedalinoMini/blob/0138b4f3c949a73c0d5f26d99fc8bcab838f8e12/src/DisplayOLED.h#L16

rarbuthnot, 2020-03-24:
Thank you @alf45tar for the quick response! I changed that address to 0x78 (how do you find these by the way?), recompiled, uploaded, restarted but still nothing on the screen. I'm wondering at this point if I received a bad screen, but I'm not sure how to test to confirm.

alf45tar, 2020-03-24:
If 0x78 does not works try 0x7A

[image]

rarbuthnot, 2020-03-24:
Ha! I didn't even look at the back! My board looks exactly like the image above with the component soldered in the same spot next to 0x78. I tried 0x7A just to see if I'd get something different, but no luck. I do have an Uno I can try to power and check it with.

alf45tar, 2020-03-24:
Wiring looks good. I suggest to avoid the resistors because they are already included in your display board (see below).

[image]

rarbuthnot, 2020-03-25:
D'oh! Turns out I didn't have the ESP32 board pushed far enough into the breadboard. Once I did that, the display came up just fine. The good news is I did learn a lot trying to troubleshoot this issue...
FWIW, I am currently using 0x3c as the address, 'nodemcu-32s' for env.board in platformio.ini.

Thank you @alf45tar for the help earlier today. I definitely plan to contribute more to this project as I continue to build it out and learn more.

closed 2020-03-25 by rarbuthnot

Wapata, 2023-06-08:
> May I suggest to change the address from 0x3c to 0x78?
> 
> https://github.com/alf45tar/PedalinoMini/blob/0138b4f3c949a73c0d5f26d99fc8bcab838f8e12/src/DisplayOLED.h#L16

Hello !
I've made this change but I still don't have anything on my (working) screen. Is there an other place to make the change for a BPI-Leaf-S3 board ?
Is it possible to have access to this parameter via the web page ?

alf45tar, 2023-06-09:
The address depends by your display and cannot be changed via the web page because cannot be changed at runtime.

alf45tar, 2023-06-09:
BPI-Leaf-S3 board has no display. It depends of which display you attached and how.

Wapata, 2023-06-09:
Yes indeed. I have the very same display as above at the 0x78 adresse,  with SDA to pin 42, SCL to pin 41 and 3.3v and ground.
I've also tried pin 15 and 16 (that look like native SDA SCL pins for this board but not as your schematics).

alf45tar, 2023-06-09:
Did you added `-D SSH1106WIRE` into `platform.ini` under `build_flags_bpi_leaf_s3`?

Wapata, 2023-06-09:
Nope...
I've seen something about "flag" during my reshearch but i've not a single clue of what it is.
Nor where it is. Are we still into Visual Studio Code / PlateformIO ?
[image]

Wapata, 2023-06-09:
Nope, it doesn't work on my side.
two hours later i've restarted from the very beginning, Changed the DisplayOled.h files to put the 0x78 adresse everywhere. Added the -D SSH1106WIRE line
Commented the ;-D SSD1306WIRE line
Build
Upload
Upload FileSystem Image
Push the firmware via https://alf45tar.github.io/PedalinoMini/installer
Disconnect reconnect the board to put the wifi password
And the web page is okay, but the display doesn't show anything.
...
I'm going to sleep now, thanks for the help anyway !:

Wapata, 2023-06-10:
Good morning,
I've changed theses line too as theses pin are unused : 
`#define OLED_I2C_SDA      40
#define OLED_I2C_SCL      39`

But still does not work. on both screens.
So, if i have to buy an other screen, do you have a recommendation ?