alf45tar/PedalinoMini #93: USB Midi issue

ciapamusche, 2020-03-24:
Hi. Thank you for sharing.

My project with Pedalino seems to be ok Except for USB midi communication.

I loaded in Arduino Pro mini the firmware USBMIDI Converter as you indicate.

I enabled USB midi from web interface. When I press the pedal, A led in my Arduino pro mini indicate the rx message incoming but in my Software (logic,amplitude) nothing arrive. Im using Mac.

should I see some kind of interface in my Midi settings or I have install drivers?

Thank you

alf45tar, 2020-03-24:
There is nothing to install on a Mac. PedalinoMini is plug & play.

[image]

alf45tar, 2020-03-24:
It would be great if you could share some photos or a video of your PedalinoMini build.

Thanks
alf45tar

ciapamusche, 2020-03-25:
Actually I have my circuit on Breadboard. I think isolate the issue.

My Arduino is a Pro Micro with USB port but the controller is ATMEGA 328P and don't have native USB support as the Arduino micro which have ATMEGA 32U4

For this reason my Mac don't recognize the usb (I think).

I read in the library page which I can use V-USB function but I cannot figure out at the moment. Have you experience?

alf45tar, 2020-03-25:
I don' have experience with V-USB but it could works.

closed 2020-03-25 by alf45tar

ciapamusche, 2020-04-01:
I used the correct Arduino Board.
Now Mac recognize the usb device but no message is received from Mac.
I used 5 V from raw pin of Arduino micro and shared same gnd.

Test with midi studio from Mac to Arduino is ok (I see blinking led).

Enebled USBMIDI from menu of Pedalino.

with I'm wrong? I m using esp32 doitdevkit v1

alf45tar, 2020-04-02:
Do not connect Vcc of your Arduino Pro Micro. It is powered by USB.
Connect only RX TX and GND as indicated into the schematic.
Did you uploaded the following sketch into Arduino Pro Micro?

https://github.com/BlokasLabs/USBMIDI/blob/master/examples/UsbMidiConverter/UsbMidiConverter.ino

ciapamusche, 2020-04-02:
yes only 3 wire from Arduinomicro and ESP32 TXD RXD and GND on D18 D19 and GND.
Tried to invert D18 D19 no success.

The ESP32 has a separate 5V coming from a switching regulator.

alf45tar, 2020-04-02:
What about the firmware to upload to the Arduino Pro Micro?
Without it there is no comunication from ESP32 and Mac.

ciapamusche, 2020-04-02:
Yes the fw is the same. Only I changed serial1 with serial because fw is wrote for leonardo

alf45tar, 2020-04-02:
It works for me without any changes.
I don't think it is correct to change Serial1 with Serial.
Serial is USB.
Serial1 is pin 0 and 1.

ciapamusche, 2020-04-02:
If not. It not compile.

ciapamusche, 2020-04-02:
Update.

Download again Library and loaded on another pc With Arduino IDE and W10. Serial1 it works and the device works correctly. The description of the sketch can Trick you because Serial1 is indicated for Arduino Leonardo. And it suggest to change it. The Serial MUST REAMAIN Serial1.

Thank you. I will update you for my project soon