alf45tar/PedalinoMini #94: Access to Web UI

bmarshall91, 2020-03-28:
I cannot connect to the web UI using pedal_name.local but can only connect using the IP address. Is there something I’m doing wrong? 

Thanks

alf45tar, 2020-03-28:
Are you using Win or Mac?
.local address is based on mdns discovery. Are PedalinoMini and your PC on the same network?
PedalinoMini is using 2.4GHZ wifi. If you PC is on 5GHz wifi mdns broadcast could not work.
Please confirm that your PC is connected to the same SSID wireless of PedalinoMini.

bmarshall91, 2020-03-29:
Computer is a Mac, running on 5GHz on the same network. I'll change it and give it another try.

alf45tar, 2020-03-30:
Connect your Mac on the same 2.4GHz wifi used by PedalinoMini and it works like a charm.

closed 2020-04-14 by alf45tar