alf45tar/PedalinoMini #96: No webserver page in normal mode

ciapamusche, 2020-04-07:
Hi, sorry for my continuous issues. 
[image]
 I finish my project, now it’s time to use it .

I play with it all the day but I notice now that I cannot reach the webserver in normal mode. I can see the pAges in wifi only. I try to erase flash reinstall but nothing to do...

Web page is only shown in wifi only mode. If I try to enable Bluetooth only at the reboot and only if I select Bluetooth only
 Some advice?

prisme60, 2020-04-07:
Hello,
+1 Same here, I have done an update to the last code, and the webserver is not more launched at startup.

ciapamusche, 2020-04-07:
Another information is that the device is correctly connected to the network. I can see the ip from display.

ciapamusche, 2020-04-07:
Issue #86 

Seems to be normal. My fault.

Alf45tar. Is possible in your opinion use the legacy midi output if not used (my case) to have extra buttons for bank+ and bank-?

alf45tar, 2020-04-08:
Yes, but only for digital input.

closed 2020-04-08 by alf45tar

ciapamusche, 2020-04-08:
I’m trying the ladder config with the configuration tool function.

It works great but I want to ask you if is normal which on press button the value goes to 127 and on release is insensitive . 

Amplitude For iPad accept only CC commands and expect for example a 0 for deactivate the effect.

Is possible in some way return to 0?  e.g press again the button?

alf45tar, 2020-04-08:
Check the connection. Tip must be connected to an analog pin and ring to a digital pin.

PedalinoMini can alternate two CC values. Use Midi value 1 and Midi value 2.

ciapamusche, 2020-04-08:
The normal momentary  In the same connection work correctly but the ladder pushbuttons give only value 1. 
I have the ladder 5 configuration .At release nothing happens.

I ll try to check again

So you confirm it can work on press and on release also on ladder?

ciapamusche, 2020-04-09:
The schematic reports tip to digital and ring to analog.

I connected as schematics . Must I invert ?

ciapamusche, 2020-04-09:
Another notice. On ladder configuration during release the bar go full scale and on press it reads the different values.

Is normal the behaviour on release ?

alf45tar, 2020-04-09:
It's depend. TC-Helicon Switch 6 requires a cross cable to connect tip to analog and ring to digital.

If the ladder is self made tip to digital and ring to analog is ok because you can connect the resistors ladder as indicated into the schematic.

On release the bar go to full scale due to the pull up resistor.

ciapamusche, 2020-04-10:
Hi confirm the value don’t return to zero on release.... I can get work ladder 5 and 6. 
Monentary function works 
Configuration tool work as expected but when I release nothing happen. 
According to midi wrench the data goes to 127 on press and don’t change on release. Every push button. I recall F1-F6 in the same profile. Select channel 1 and different CC  for every pushbutton. MIDI value 1 to 127. MIDI value 2 to 0

Connected as schematic