alf45tar/PedalinoMini #97: Ladder issues. No autosense. No cc to release

ciapamusche, 2020-04-11:
Hi... Thank you again for yor patience Alf45tar.
Today i performed some test on my project where i had issues on ladder configuration.

The problem is: The pushbuttons are insensitive to release.

Actions:
-Connect a ladder as schematic on pedal 6 on Profile B
-Set a ladder on pedal 6  as picture with autosense enable and latch on 1 and 2
[image]

-On profile B1 Renamed the F channel F1 to F6 and sent to ch1 control 80 to 86.
-F1 to F4 have Value one to 127 and Value 2 to 0
-F4 to F6 have Value 1 to 0 and Value 1 to 127
-Perform autocalibration from main menu succesfully
-Connected the monitor
-Recall profile with pushbutton to visualize the data in terminal
- Try to push every button in sequence 
-Try to push Pushbutton 1 in control 60  and 2 in control 61. (they return to 0)

Output:
Boot WIFI
USB MIDI started
DIN MIDI started
Connecting to
SSID        : XXX
Password    :XXX
SYSTEM_EVENT_WIFI_READY
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_STA_CONNECTED
BSSID       : B0:39:56:F2:F5:AD
RSSI        : -77 dBm
Channel     : 2
STA         : 24:6F:28:B4:5E:20
SYSTEM_EVENT_STA_GOT_IP
Hostname    : B4286F24
IP address  : 192.168.1.173
Subnet mask : 255.255.255.0
Gataway IP  : 192.168.1.1
DNS 1       : 8.8.8.8
DNS 2       : 8.8.4.4
mDNS responder started
OTA update started
SPIFFS mount OK
HTTP server started
Connect to http://B4286F24.local/update for firmware update
Connect to http://B4286F24.local for configuration
ipMIDI server started
RTP-MIDI started
OSC server started
Internal Total Heap 254844, Internal Free Heap 132268
Loading profile ...
Reading NVS Profile B ... done
Pedal autosensing...
Pedal  1   autosensing disabled
Pedal  2   autosensing disabled
Pedal  3   autosensing disabled
Pedal  4   autosensing disabled
Pedal  5   autosensing disabled
Pedal  6   autosensing disabled
Bank  1
Pedal  1     MIDI         LATCH1       PRESS_1       POLARITY+   CONTROL_CHANGE      60   Channel  1   Pin D25
Pedal  2     MIDI         LATCH1       PRESS_1       POLARITY+   CONTROL_CHANGE      62   Channel  1   Pin D26
Pedal  3     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         64   Channel  1   Pin D27
Pedal  4     MIDI         MOMENTARY1   PRESS_1       POLARITY+   NOTE_ON_OFF         65   Channel  1   Pin D14
Pedal  5     MIDI         MOMENTARY1   PRESS_1       POLARITY+   CONTROL_CHANGE      12   Channel  1   Pin D12
Pedal  6     MIDI         LADDER       PRESS_1       POLARITY+   CONTROL_CHANGE      80   Channel  1   Pin A33
Pedal  2   input 1 output 1
MTC None
Pedal  5   input 1 output 1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  80......Value 127.....Channel  1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  81......Value 127.....Channel  1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  82......Value 127.....Channel  1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  84......Value 127.....Channel  1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  85......Value   0.....Channel  1
Pedal  6   SINGLE PRESS
CONTROL CHANGE.....Code  86......Value   0.....Channel  1
Pedal  1   input 0 output 0
CONTROL CHANGE.....Code  60......Value 100.....Channel  1
Pedal  1   input 1 output 1
CONTROL CHANGE.....Code  60......Value   0.....Channel  1
Pedal  2   input 0 output 0
CONTROL CHANGE.....Code  62......Value 100.....Channel  1
Pedal  2   input 1 output 1
CONTROL CHANGE.....Code  62......Value   0.....Channel  1

As yo note the ladder go only to Value 1 

Another thing. The autosense flag is ignored.
any ideas?
Thank you very much .

alf45tar, 2020-04-14:
Alternation between CC midi value 1 and 2 is only available for momentary, latch and latch 2 pedal mode.
Momentary 2, momentary 3 and ladder do not support it yet.

ciapamusche, 2020-04-15:
Thank you for the reply ... 

I think it can work also only if return automatically fixed to zero after some millisecond. 

I tried on AmpliTube for iPad. 

I think You can use it for global buttons (eg. The looper) if it returns to Zero. 

Thank you again

alf45tar, 2020-07-06:
Check the latest version 2.0 just released

closed 2020-07-06 by alf45tar