alf45tar/PedalinoMini #98: OLED display with Adafruit Huzzah32

wrpooley, 2020-04-28:
Greetings.  I'm having some success putting this project on an Adafruit Huzzah32 but I'm having no luck with an OLED display.  Everything works great with a DevKit board.  I am able to make the display work with the Huzzah32 by loading the example code attached from Adafruit.

I'm new at this so I'm sure I'm missing something elemental.  I'm poking around the DisplayOLED.h code for discrepencies though I'm not what I'm looking at!

Thanks in advance for any help any of you might offer

wp

[ssd1306_128x32_i2c.txt](https://github.com/alf45tar/PedalinoMini/files/4548058/ssd1306_128x32_i2c.txt)

alf45tar, 2020-04-29:
Did you try to define SSD1306WIRE and change

#define OLED_I2C_ADDRESS  0x3d
#define OLED_I2C_SDA      23
#define OLED_I2C_SCL      22

wrpooley, 2020-04-29:
I had not yet modified DisplayOLED.h.  I did however make it work by doing this:

`#if defined(SSD1306WIRE)
#include <SSD1306Wire.h>
#define SSD1306WIRE            
#define OLED_I2C_ADDRESS 0x3c
#define OLED_I2C_SDA 23
#define OLED_I2C_SCL 22
SSD1306Wire               display(OLED_I2C_ADDRESS, OLED_I2C_SDA, OLED_I2C_SCL);
#endif`

Adding #define SSD1306WIRE didn't make it work but it also didn't make it NOT work!

I believe it was the 23 and 22 define.  How did you know it was those numbers?  Also, 0x3d did not work but 0x3c did however.

Thanks as always for your help!

wrpooley, 2020-04-29:
Oh hell!  It says GPIO 23 and 22 on the back.  Was I on the right track with:
#define SSD1306WIRE?  It seemed to throw up a problem in VSCode about it being "redefined"

closed 2020-04-30 by wrpooley