alf45tar/PedalinoMini #99: esp resets on boot (PSRAM not found)

f0m3, 2020-05-02:
Hello,
i managed to compile pedalino for my esp32. But it doesnt boot normally. In the boot process it resets itself in an endless loop. Any suggestions why? Heres the serial log:

```
Reset EEPROM to factory default
Updating NVS ... done
[NVS][Global][Device Name]: CBAB6224
Updating NVS ... done
[NVS][Global][Boot Mode]: 3
Updating NVS ... done
[NVS][Global][STA SSID]:     myownssid
[NVS][Global][STA Password]: myownpwd
Updating NVS ... done
[NVS][Global][AP SSID]:     Pedalino-CBAB6224
[NVS][Global][AP Password]: CBAB6224
Updating NVS ... done
[NVS][Global][Bootstrap Theme]: bootstrap
Updating NVS ... done
[NVS][Global][Current Profile]: 0
Updating NVS ... done
[NVS][Global[Tap Dance Mode]: 0
Updating NVS ... done
[NVS][Global[Bank Switch]: 0
Updating NVS ... done
[NVS][Global[Single Time]: 200
[NVS][Global[Double Time]: 400
[NVS][Global[Long   Time]: 500
[NVS][Global[Repeat Time]: 500
Updating NVS ... done
[NVS][Global][Ladder]:
Ladder 0 Threshold 497 Tolerance 82
Ladder 1 Threshold 660 Tolerance 47
Ladder 2 Threshold 752 Tolerance 33
Ladder 3 Threshold 816 Tolerance 31
Ladder 4 Threshold 876 Tolerance 31
Ladder 5 Threshold 945 Tolerance 35
Updating NVS ... done
[NVS][Global[Encoder Sensit]: 5
Updating NVS ... done
[NVS][Global[Blynk Cloud]: 0
Updating NVS ... done
[NVS][Global][Blynk Token]: 
Updating NVS Profile A ... done
Updating NVS Profile B ... done
Updating NVS Profile C ... done
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 238260, Internal Free Heap 209892
PSRAM Total Heap 0, PSRAM Free Heap 0
PlatformIO Built Environment: ttgo-t-eight
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___ 
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \ 
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  / 
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star      
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: CBAB6224
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 253628, Internal Free Heap 225144
PSRAM Total Heap 0, PSRAM Free Heap 0
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      CBAB6224
[NVS][Global][Boot Mode]:        3
[NVS][Global][STA SSID]:         myownssid
[NVS][Global][STA Password]:     myownpwd
[NVS][Global][AP SSID]:          Pedalino-CBAB6224
[NVS][Global][AP Password]:      CBAB6224
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][Blynk Cloud]:      0
[NVS][Global][Blynk Token]:                                      
[NVS][Global][Ladder]:           Ladder 0 Threshold 497 Tolerance 82
[NVS][Global][Ladder]:           Ladder 1 Threshold 660 Tolerance 47
[NVS][Global][Ladder]:           Ladder 2 Threshold 752 Tolerance 33
[NVS][Global][Ladder]:           Ladder 3 Threshold 816 Tolerance 31
[NVS][Global][Ladder]:           Ladder 4 Threshold 876 Tolerance 31
[NVS][Global][Ladder]:           Ladder 5 Threshold 945 Tolerance 35
###############################################################################################################################################################
```

alf45tar, 2020-05-02:
Which ESP32 board do you have?
You uploaded PedalinoMini for ttgo-t-eight.

f0m3, 2020-05-02:
Hmm ordered it at ebay... Maybe its a wrong one. I did a lot ESP8266 but this is my first ESP32. It is labeled TTGO: [Bangood](https://www.banggood.com/LILYGO-TTGO-T-Display-ESP32-CP2104-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-p-1522925.html?rmmds=myorder&cur_warehouse=UK)
Is it the wrong module?

alf45tar, 2020-05-02:
Yes, it is. TTGO-Display cannot work with PedalinoMini because the display ST7789V is not supported.

closed 2020-05-07 by alf45tar