alf45tar/PedalinoMini #102: After the completion of the upload, step 5 and 6

zhanwu123, 2020-05-28:
> Executing task in folder PedalinoMini: C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
2:57

rst:ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 329204, Internal Free Heap 300416
PSRAM Total Heap 0, PSRAM Free Heap 0
PlatformIO Built Environment: esp32doit-devkit-v1-wifi
_________           .___      .__  .__                 _____  .__       .__
 ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|
/  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (
 (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/
\__\                 \/  /__/

   (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 60C40A24
PSRAM not found
Internal Total Heap 329204, Internal Free Heap 300416
PSRAM Total Heap 0, PSRAM Free Heap 0
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      60C40A24
[NVS][Global][Boot Mode]:        2
[NVS][Global][STA SSID]:
[NVS][Global][STA Password]:
[NVS][Global][AP SSID]:          Pedalino-60C40A24
[NVS][Global][AP Password]:      60C40A24
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][Blynk Cloud]:      0
[NVS][Global][Blynk Token]:
[NVS][Global][Ladder]:           Ladder 0 Threshold 497 Tolerance 82
[NVS][Global][Ladder]:           Ladder 1 Threshold 660 Tolerance 47
[NVS][Global][Ladder]:           Ladder 2 Threshold 752 Tolerance 33
[NVS][Global][Ladder]:           Ladder 3 Threshold 816 Tolerance 31
[NVS][Global][Ladder]:           Ladder 4 Threshold 876 Tolerance 31
[NVS][Global][Ladder]:           Ladder 5 Threshold 945 Tolerance 35

Boot BLE
USB MIDI started
DIN MIDI started
Internal Total Heap 329012, Internal Free Heap 299308
Loading profile ...
Reading NVS Profile A ... done
Pedal autosensing...
Pedal  1   autosensing disabled
Pedal  2   autosensing disabled
Pedal  3   autosensing disabled
Pedal  4   autosensing disabled
Pedal  5   Tip Pin 12 Value  680    Ring Pin 32 Value  901 ANALOG POLARITY+
Pedal  6   Tip Pin 13 Value  570    Ring Pin 33 Value  871 ANALOG POLARITY+
[... 198 lines cut ...]
Pedal  5   input 164 output 37 velocity -604.17
CONTROL CHANGE.....Code  12......Value  37.....Channel  1
CONTROL CHANGE.....Code  12......Value  37.....Channel  1
Pedal  5   input 153 output 44 velocity 1.52
CONTROL CHANGE.....Code  12......Value  44.....Channel  1
CONTROL CHANGE.....Code  12......Value  44.....Channel  1
Pedal  5   input 0 output 127 velocity 512.35
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Pedal  6   input 60 output 0 velocity -1.32
CONTROL CHANGE.....Code  13......Value   0.....Channel  1
CONTROL CHANGE.....Code  13......Value   0.....Channel  1
Pedal  5   input 180 output 24 velocity -5.80
CONTROL CHANGE.....Code  12......Value  24.....Channel  1
CONTROL CHANGE.....Code  12......Value  24.....Channel  1
Pedal  6   input 0 output 127 velocity 9.01
CONTROL CHANGE.....Code  13......Value 127.....Channel  1
CONTROL CHANGE.....Code  13......Value 127.....Channel  1
Pedal  5   input 24 output 127 velocity 6.31
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
CONTROL CHANGE.....Code  12......Value 127.....Channel  1
Pedal  6   input 76 output 0 velocity -9.07
CONTROL CHANGE.....Code  13......Value   0.....Channel  1
CONTROL CHANGE.....Code  13......Value   0.....Channel  1
Pedal  6   input 0 output 127 velocity 52.59

alf45tar, 2020-05-28:
Sorry, which is the question or issue?

zhanwu123, 2020-05-30:
https://v.youku.com/v_show/id_XNDY5MTg2NDQ2OA==.html

alf45tar, 2020-05-30:
I suggest to have a look of issue https://github.com/alf45tar/PedalinoMini/issues/59

closed 2020-07-06 by alf45tar