alf45tar/PedalinoMini #103: WIFI set up after the restart

zhanwu123, 2020-06-02:
https://v.youku.com/v_show/id_XNDY5NTQ1NzQ2MA==.html

zhanwu123, 2020-06-02:
Boot mode 3 and 5 cannot be used

alf45tar, 2020-06-03:
It could be helpful to have the log during boot.

zhanwu123, 2020-06-03:
WIFI can be connected, but can't find the bluetooth

alf45tar, 2020-06-03:
According to your log on issue #102 you upload the wifi only version esp32doit-devkit-v1-wifi.
Upload esp32doit-devkit-v1-ble or esp32doit-devkit-v1 (wifi+ble).

zhanwu123, 2020-06-03:
> Executing task: C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 241644, Internal Free Heap 213220
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 20480
PlatformIO Built Environment: esp32doit-devkit-v1
_________           .___      .__  .__                 ____

zhanwu123, 2020-06-03:
This version is right

alf45tar, 2020-06-04:
Yes, it is. Just select a boot mode which includes BLE.

zhanwu123, 2020-06-04:
Version 1 WIFI and bluetooth can run normally, version 2.0 bluetooth is normal, WIFI cannot be used alone

zhanwu123, 2020-06-04:
IK AmpliTube app bluetooth even without other app can be normal use

zhanwu123, 2020-06-04:
IK AmpliTube app bluetooth even don't, don't know what's the problem

mknerr, 2020-06-04:
The app needs a facility to connect to a Bluetooth device. I’m sure Amplitube will respond to midi, but it may not have the setting to connect to Bluetooth.

zhanwu123, 2020-06-04:
I didn't use it, I use a IPHONE6

zhanwu123, 2020-06-04:
I didn't use IPAD, I use a IPHONE6

alf45tar, 2020-06-04:
Which is the difference? They are both iOS devices. It should be pretty the same.

zhanwu123, 2020-06-07:
https://v.youku.com/v_show/id_XNDcwMjc0MDMyNA==.html
This is for IPHONE Amplitube

alf45tar, 2020-07-06:
Do not use iRig Blueboard because it is using a proprietary protocol.
Connect PedalinoMini via bluetooth using midimittr app or GarageBand.

closed 2020-07-06 by alf45tar