alf45tar/PedalinoMini #104: Status LEDs enhancement

aFunkyBass, 2020-06-06:
Hello there, first of all, thank you for your great job!

I'm about to build a Pedalino Mini based foot controller to make it work with Blokaslabs MODEP. In order to be really suitable for live performing, it's very important to have a visual feedback of switches status. 

I saw you implemented a shift-register system for status LEDs, but I saw that it's rather limited. Have you considered going for the so-called neopixels (WS2812b)? This could be a nice improvement  as it allow to have more flexibility in assigning LEDs to footswitches adding the possibility to choose colours, intensity and so on.

Let me know your thoughts.

alf45tar, 2020-06-07:
Thanks for the hint. Neopixels looks very interesting even if not so cheap.

jimb-cc, 2020-07-15:
Hi I'd like to second this enhancement.  I've been working on a 3D printable pedalboard which will also use WS2812b addressable LEDs for status, The pedal is designed to be printed with a translucent filament, so the whole thing will glow to indicate state.  You are right that the addressable LEDs are more expensive than the basic LEDs, but at less then $2 for 10, the cost isn't prohibitive.  As I get more familiar with Pedalino, I might have a go at this myself, but I'm still very much on the physical design right now!  Thanks again for the amazing project! 
[image]

for [example](https://www.ebay.co.uk/itm/10-50-100PCS-Mini-WS2812B-WS2811-IC-10mm-3mm-DC5V-SMD-5050-RGB-Led-light-Sample/283586408298?hash=item4207112f6a:g:wSwAAOSweyBdW8pl):

aFunkyBass, 2020-07-15:
Hey @jimb-cc, I'm happy that you second this.

As I'm building a pedalboard with more than 6 switches, the shift-register system isn't enough for the purpose. Just today I was going through the code to see if I could write a module myself, but that would require to hack several things and it would be way easier for @alf45tar than for me... ;)

Not having limitation on LED number, you can assign a single LED for every switch, even if you have 6 ladders, it's not a problem to handle 36 LEDs... Not only that but, it would be great if you could set separate colours for every switch/state on the web UI. Say on pedal 1 you have assigned a tube screamer (which is traditionally green after Ibanez TS9), you can give a dim green on the off state, and a bright green when it's on. You could even assign fading colours for expression pedals, following the values from min to max.

aFunkyBass, 2020-07-20:
> They are no so expensive consider the total cost of a PedalinoMini. I just order some of them. I think they will replace the standard leds.

Awesome! Thank you @alf45tar

alf45tar, 2020-08-02:
Added support for WS2812B RGB led strip. @aFunkyBass and @jimb-cc if you want to give a try and let me know.

aFunkyBass, 2020-08-02:
Thanks @alf45tar! 
This week I''m going to update the code and try it! :wink:

jimb-cc, 2020-08-02:
Great work, really looking forward to this, I'll test it out on my
breadboard hopefully tomorrow!

Can you give us a quick heads up how to use the new functionality?  If I
get time I'd be happy to send of a PR for an update the docs once I've got
it wired up.

Jim

alf45tar, 2020-08-03:
Connect your led strip to Vin (5V), GND and GPIO_NUM_5 (Din).
For every Actions select the led and the color to power on.

I will amend the documentation once tested.

jimb-cc, 2020-08-03:
alf45tar,
Thanks for the amazing work getting the WS2812 support in so quickly, it
looks amazing.  I've got it running on my test rig here and looks so pretty!

Anyway, I seem to have an issue with Momentary vs. Latch, but it could just
be the way I have things set up,

I've attached my CFG file to this email in case it helps.

My test rig is based on a Hornbill ESP32 Devboard and has 3 microswitches
and two pots. Excuse the messy wiring, I'll run proper jumpers in later
when it's fairly fixed down.

I've recorded a quick video to just talk through how it looks, let me know
if I can provide any more detail, I'm keen to support any testing that you
want to do.

https://photos.app.goo.gl/Tc6aZcSHt42XrTnr9

[image: image.png]

[image: image.png]
[image: image.png]

raidolo, 2020-08-04:
Hi @alf45tar, your work is amazing! 

I've found a strange behaviour of the leds. 

I'm using 3PDT latch pedal buttons for my project and ws2812 addressable leds.
When I set the led color on release, using the same LED number, the software just light up the led on release, and no more on press. 

Is this to be expected?

Another thing:
The app I'm controlling is GE LABS on iOS. It has 10+2 buttons (see the image) 

[image]

Excluding the bank switches on the left, the app has 5 buttons to toggle the single effects in the preset, and 5 buttons to switch the presets within the current bank. The thing is: the top 5 buttons are fine with latch led behavior, because I want to turn on or turn off the single effect pedal, and when I do it I want to turn on the correspondent led (and turn off or change color on release, so I'm expecting this behavior for the problem reported above). 
The preset buttons in the row below, are just actionable one time, you cannot "turn off" the current preset, you can just switch to another one, so the latch behavior don't match and I had to issue the same Midi Control Change on "press" and on "release", because basically pressing again the same latch button, whatever it's state is, should just activate the related preset, and also the led behavior don't match because if I press the pedal for another preset, no one is "releasing" the previous pedal and the led of the previous one stays on. I know I could have used a momentary switch, but the "click" action is not that fast compared to a press or a release, so using latch buttons makes the actions just faster. Also in the case of a momentary switch, no actions will turn off the led of the previously clicked button, so if I switch the 5 presets I end up with 5 leds on.
I'm also planning to use a total of 10 latch buttons, using the "click" additional action on two of them to switch the banks within the app (it works fine even with latch buttons, already tested). 

This long preamble to say that, using addressable leds could open to a lot of possibilities, 
I'm imaging for instance a matrix for each actions, where sending a midi command, could set the whole addressable led matrix, not just a single led, or use neopixels special effects such as led flashing. I've seen something like this in projects made for flying drones.
Imagine for instance I'm switching banks using the click action as I've described above, I'd like to flash the entire strip with white color...
This way you can decouple the LED behavior from the physical push buttons behavior and relate them to the "command" sent. 

Why? Because you know it would be cool!

Cheers,
Alessio.

mknerr, 2020-08-05:
This is a fantastic addition! Great work, as usual. 

Just to add a bit to what @jimb-cc was saying in the video about a minimum and maximum color, I was also thinking it would be nice to have an intensity setting for the "off" state, as it is customary to color code the switches as to the type of effect you are switching, and have a visual reminder of what you have programmed.

alf45tar, 2020-08-05:
I am inviting you to consider the implementation of WS2812B addressable leds just started. More to come.

alf45tar, 2020-08-05:
Recent progress:
- double colors
- brightness

raidolo, 2020-08-05:
> Recent progress:
> 
> * double colors
> * brightness

That was fast! I’ll test it soon. Thanks!

raidolo, 2020-08-06:
Hi @alf45tar 

I've compiled the latest master.
I don't get the double tag option and the double color. 

Could you explain a bit how to use them? 

Using a latch button, to get the right behavior of a single led (color 1 on press, color 2 on release action), I had to put the same color on both colors for press action and the other color on both colors on the release. 
Let's imagine I want red for press and green for release, I had to configure two reds on press and two greens on release. 
I tried with 4 different colors but after the first cycle, only the secondary colors show up.
I was expecting 4 colors first press, first release, second press, second release and so on....

Also, I was not able to show the "off" tag, it will never appear. If set only the "off" tag no tag appears at all, just the code sent.

The led brightness works just fine.

Thanks!
Alessio.

alf45tar, 2020-08-06:
I know it is not perfect. I am working on it.

raidolo, 2020-08-06:
> I know it is not perfect. I am working on it.

Don't take my comments as "to point the finger" on errors, but rather than as a user experience feedback. 
Let me know if I can help, other than test the future releases...

akosbeke, 2020-08-07:
Hi there,

Another consideration could be the use of a per-led brightness for the ON and OFF states, meaning the pedal in an OFF state would be the dimmer version of the color and if it's ON then it would be brighter.
Much like on a Helix.

I see you are taking advantage of the FastLED library. In this case a straight forward implementation could be like this:
```cpp
#include <FastLED.h>

const LED_NUMBER = 2;
const LED_PIN = 5;
const ON_BRIGHTNESS = 255;
const OFF_BRIGHTNESS = 50;

CRGB leds[LED_NUMBER];

FastLED.addLeds<NEOPIXEL, LED_PIN>(leds, LED_NUMBER);

void setup() {
    leds[0] = CHSV(0, 255, ON_BRIGHTNESS);   // red
    leds[1] = CHSV(160, 255, ON_BRIGHTNESS); // blue
}

void loop() {    
    // Turn on
    for (int i = 0; i < LED_NUMBER - 1; i++)
    {
      CHSV fullBrightnessColor = leds[i];
      fullBrightnessColor.v = ON_BRIGHTNESS;
      leds[i] = fullBrightnessColor;
    }

    FastLED.show();
    delay(1000);

    // Turn off
    for (int i = 0; i < LED_NUMBER - 1; i++)
    {
      CHSV dimmedColor = leds[i];
      dimmedColor.v = OFF_BRIGHTNESS;
      leds[i] = dimmedColor;
    }

    FastLED.show();
    delay(1000);
}
```

Have a great day!

akosbeke, 2020-08-10:
I saw and tested the brightness feature. It works great.

One more LED related thing came to my mind: it would be awesome to have a blinking LED for the tap tempo pedal for a visual representation of the current BPM. (Again: much like on a Helix)

alf45tar, 2020-08-11:
@akosbeke define a Tap tempo action and the relative led to test the new feature.

akosbeke, 2020-08-12:
I'm afraid I can't make the blinking work.

I tried with separate and global TAP actions but couldn't make the LED blink.

I could have messed up something though.

alf45tar, 2020-08-12:
Do not define TAP at pedal level. Define a new TAP action and select 2 different color and/or 2 brightness (in Options).

akosbeke, 2020-08-12:
I tried that too, but no luck. The monitor only says: “Action: “. It feels like the tap tempo feature does not work for me.

Am I missing something?

alf45tar, 2020-08-12:
Sorry you are right. You have to activate MIDI Clock as master. At the moment the only way is to use the /live page.

akosbeke, 2020-08-13:
I got it. Now it is working, but the live view crashes all the time, but I saw in a different issue that it is to be expected at this point.

I am testing the LED feature in depth and realized that in case of a number of PC actions on the same channel there should be really only one LED in the 'ON' status since you cannot be in multiple programs at the same time (you can either be in program number 1 or program number 2, but not both). Please give this a thought.

alf45tar, 2020-08-13:
@akosbeke if you want give a try. Only 1 PC led active at the same time on the same channel.

akosbeke, 2020-08-13:
Wow, thanks. It's working well.

Is there a way to set a particular PC number to be default so the LED is in the 'ON' state after boot?

I have the same question about CC messages. Can I set an action to be 'ON' by default so the first press of the button actually turns it off?

Sorry for the constant brainstorming. :) I just got super excited about the LED feature because it really makes a difference on gigs.

alf45tar, 2020-08-14:
@akosbeke thank for the suggestion.
Add a dot at the end of tag to set the initial color of the led.
Adding a . at the end of the tag on the led will be set with the on color.
Adding a . at the end of the tag off the led will be set with the off color.
By default (if no . are present) the off color will be used.

In case up multiple action assignment to the same led the initial led color is determined by the last . sign.

akosbeke, 2020-08-14:
@alf45tar While the default LED status work perfectly, this kind of introduces a new bug.

This way, if I choose a CC action to be on by default then if I press the button the LED will turn off (as expected) but the CC value of 127 will be sent. 
A viable solution could be to check the existence of the dot in the on/off tag and in case of a CC message, send the value accordingly.

In case of PC actions it works as is.

alf45tar, 2020-08-14:
@akosbeke bug fixed

akosbeke, 2020-08-14:
@alf45tar awesome, it works great. thanks

zhanwu123, 2020-09-09:
Hello, WS2812B lamp belt is a + 5 v, connecting ESP32 GND and 5 feet

alf45tar, 2020-09-16:
WS2812B lamp belt working also with 3.3V

jimb-cc, 2020-09-16:
Just an update, I have LEDs in my pedals now :)
https://photos.app.goo.gl/dfprKvhPtvy1kPqB6

alf45tar, 2020-09-16:
@jimb-cc did you printed the pedal? Do you have the 3D model to share?

jimb-cc, 2020-09-16:
yes, 3D printed, it's not quite finished yet I'm still working on it, but
It's designed specifically to work with Pedalino, and I'll make sure you
get to see it when it's done :)

Jim

rigr, 2020-09-22:
New to neopixels I'm very intzerested to gain some knowledge.
So, please help me to understand:
Can someone point out how to connect them? Is there a wiring diagram I could follow?
 Thanks and best regards!

akosbeke, 2020-09-22:
Hey @rigr you're gonna have to connect your neopixel's data pin to GPIO_NUM_5, GND to the common ground rail and the +5V to Vin (5V).

After that set a LED number and color for you actions.

rigr, 2020-09-22:
Thanks @akosbeke, will try that soon.

rigr, 2020-09-23:
Hello, first of all again thanks to alf45tar for this great project.
I hope this is the right place for some brainstorming regarding neopixels. Please read the rest as ideas - in fact I'm very humble and would not want anybody to read this as "please make it this way".
1) depending on what is connected the neopixels should show different behavior: 
  a) Using a pedal reading an analog value  (Volume etc.) it would feel "natural" to me that color changes from dark to bright color, 
  b) Using a momentary switch to select between one, double or long click neopixels should change to pre-defined colors, so I see what state the pedal is at.
2) When tapping a certain speed, I would feel ok, that either the according or all neopixels show the rhythm with a change of the brightness
3) Setting the pedalino to use neopixels or LEDs should be easy to set (or even better: would be great if it was discovered by the ESP)
4) when switching on or after a reboot I would LOVE to have all neopixels glow in a dark red color, just to say hello and to show that all is connected correctly. 
5) while pressing a button the according neopixel should show activity by glowing a bit brighter (or if already full bright in a little different color).
That’s it for the moment. 😊
Thanks and best regards!