alf45tar/PedalinoMini #105: Minimal firmware

mknerr, 2020-06-06:
Love seeing all the movement on this project. I’m still tinkering when I get a chance. I had an idea and wondered if it was feasible, and if so, the difficulty involved.

Could PedaliniMini be minimized enough to be installed on a esp-01? The thought would be to have a simple WiFi and/or BluetoothLE transceiver. It would just have MIDI in/out, or possibly implement USB midi as an alternative. Ideally, it would be powered via the MIDI DIN jack or the USB port. 

Thoughts?

alf45tar, 2020-06-07:
I don't think so. ESP8266 has no bluetooth, not enough memory and not enough IO pins without external expansion.

You could do something with limited features. But the question is: why?

mknerr, 2020-06-09:
Yeah, I had forgotten about the Bluetooth issue, but IPmidi would suffice. 

> But the question is: why [limit it]?

That had crossed my mind too, but the end goal would be to make it extremely compact, and hopefully power off of the MIDI jack and function similar to one of these:

https://usa.yamaha.com/products/music_production/accessories/md-bt01/index.ohtml

IPmidi would be even better, as it wouldn’t require any pairing, but the extra power draw would likely prevent powering off midi.

closed 2020-07-06 by mknerr