alf45tar/PedalinoMini #108: Change OLED pins

jaylfc, 2020-06-14:
Hi,

Where would you change the pin outs for an OLED screen? I’ve been battling trying to get it working on a Heltec board and the pins for the integrated screen are different. If these could be found and changed easily it could enable better support across the multitude of devices that exist with integrated displays.

Thanks!

alf45tar, 2020-06-14:
https://github.com/alf45tar/PedalinoMini/blob/3d8b40b1b096baa8616807b8f1e2b3490272fe40/src/DisplayOLED.h#L24

mknerr, 2020-06-15:
You might check out my post, #8 , from a while back. Alf45star was able to get me pretty far, though I never did get it working quite right. The display would be off by a random number of vertical pixels each time it booted. There is another person trying to get it going too. Would be nice, since it has the charging built in as well, and it's nice and compact.

jaylfc, 2020-06-16:
Hi and thanks for trying to help.

I have tried steps outlined in older posts but the screen never wakes up.
The pin outs on this board are a little different, as you can see in this diagram with the pins used by the display indicated by a red arrow:

https://www.weargenius.in/wp-content/uploads/2018/12/df.png

It would be awesome to get this board supported as I have already built a little matchbox sized wooden enclosure with battery and buttons, I'll perservere but any help would be much appreciated!

Thanks again.

alf45tar, 2020-08-04:
That board cannot be officially supported because to much pins are used by the board itself.

closed 2020-08-04 by alf45tar