alf45tar/PedalinoMini #110: Web UI various problems

aFunkyBass, 2020-07-13:
Hello there, I have some problems with Web UI. 

**First** when I enter the web configuration, it doesn't load the bootstrap module, so the interface is totally screwd up, here's what I got:

[image]

Going on the options page and changing the theme solves the problem, but it needs Internet connection

**Second** going on the  Actions page the "New Action" button doesn't do anything. I tried with several browsers under Linux, but had no luck. The only way for me to add an action is to use the Enter key when in the "Bank Name" field, but it only adds actions on Pedal 1

What am I doing wrong?

Thank you!

alf45tar, 2020-07-13:
Run “Upload File System image” task in PlatformIO IDE or use PlatformIO Core (CLI) and 'platformio run --target uploadfs' command

aFunkyBass, 2020-07-13:
Ops... I can't believe I was so dumb!! :sob: :smile: 
Sorry!

aFunkyBass, 2020-07-14:
Obviously it works flawlessly! :wink: 

Now  I have another problem, I'll put it here (even if it's OT), I can open a new issue if you prefer.

I'm building a pedalboard with 8 footswitch (momentary), so I'm using 4 channels in momentary/2 configuration, leaving 2 channels for external epression pedals.

The thing is that momentary/2 acts weird. The first button toggles values (press once sends 127, press again sends 0), which is what I need, but the second button of the momentary, always sends 127. plus it seems to affect the value of the first button. 

Here's my configuration
[image]

Is it an intended behaviour?

alf45tar, 2020-07-14:
It is not an intended behaviour. It is a bug. I tried to fix it on the fly but I didn't tested yet. Check the latest commit if it works now.

aFunkyBass, 2020-07-14:
It works as expected now, I tested it only on channel 1 though, but I'm confident about the rest of channels! ;)

Thanks a lot!

alf45tar, 2020-07-14:
Thanks to you for debugging my software

closed 2020-07-14 by alf45tar

WernAir77, 2020-12-01:
I have the same issues with the web interface. 
Which theme and which browser is working?

alf45tar, 2020-12-01:
Did you upload the file system image?

Under PlatformIO Project Tasks select your environment (i.e. env:esp32doit-devkit-v1)
Click “Upload File System Image” under Platform 

The standard system image includes only the "bootstrap" theme. For the other themes an internet connection is required.
Any modern browser is supported including mobile devices.

WernAir77, 2020-12-01:
I reinstalled the image and now it works fine. 
Merci