alf45tar/PedalinoMini #112: Multi button pedals not working

mknerr, 2020-07-17:
Finally got a chance to try out V2, but for some reason, Momentary 3, and Ladder are not working. My ladder is the TCH switch6 and I have reversed the tip and ring. I have a simple button attached to pedal 2 and it is working as expected. Just tried a new profile to start from scratch, and still nothing from Ladder6 or Mom3. Went through the updated read-me and didn't see anything I'm missing, but there must be something.

alf45tar, 2020-07-17:
Did you calibrated the ladder?
I will check Momentary 3.

mknerr, 2020-07-17:
> Did you calibrated the ladder?
I did not. For some reason I thought this was not necessary now. Will look into that.
> I will check Momentary 3.
Thank you!

mknerr, 2020-07-17:
Just a quick update; I calibrated the Ladder sw but it did not appear to have an effect.
I also tested Momentary2 which did work with my Mom3 switch, but only two controls, of course. I immediately changed it to Mom3 and it ceased working.

alf45tar, 2020-07-18:
Momentary 3 return to work. Working on Ladder.

alf45tar, 2020-07-18:
Ladder return to work.

mknerr, 2020-07-18:
Great, thank you!

mknerr, 2020-07-18:
Still having some issues with Ladder. What should the Analog curve and min/max be, in the Pedals tab, or does it matter? I've recalibrated. The response I'm getting is; no response on button 1, then my button one function occurs when I press pedals 2-6. Could still be something on my end, but I'm not sure what.

alf45tar, 2020-07-19:
Can you retry with the latest version?
If it works can you give me your calibration levels of an original TC pedal.

Thanks

mknerr, 2020-07-20:
I am getting the same result as above.
Here are my calibration levels, for what it's worth. 

Level 1
961
Level 2
888
Level 3
829
Level 4
758
Level 5
669
Level 6
500
Level 7
1023

alf45tar, 2020-07-20:
Can you calibrate it pressing pedals in the opposite order?
Levels should increase values.
Level 1 < Level 2 < Level 3 < Level 4 < Level 5 < Level 6 < Level 7

mknerr, 2020-07-20:
Okay, that did the trick, except the button numbers are backwards, of course. I can easily work around that, if need be. 

Thank you!

alf45tar, 2020-07-20:
I will implement levels sorting so pedals can be pressed in any order.

closed 2020-07-20 by alf45tar