alf45tar/PedalinoMini #113: Issues after update to V2

faierserio, 2020-07-18:
After update my PedalinoMini to V2, I have encounterd these issues: 
1) Momentary3 pedal doesn't work (nothing event on action, it works in V1) 
2) Analog pedal, if I set control change value min=0 max=64 , on display I read values moves from 0 to 64 but on midi sends values from 0 to 127 (works in V1). 
3) The last strange thing, is that Legacy Midi output doesn't working after update to V2 (what has changed?)

alf45tar, 2020-07-18:
1 and 2 are fixed in the latest version.

alf45tar, 2020-07-18:
Did you enabled the Legacy MIDI in the Interface tab?

faierserio, 2020-07-19:
Yes, I have enabled legacy midi output, also I verified that going back to version v1 it works

alf45tar, 2020-07-19:
Give a try to latest version. It should fix Legacy MIDI.

faierserio, 2020-07-26:
Hi alf45tar, I tried the latest version, the Momentary3 and Analog issues are solved, but legacy midi still doesn't works

rafm44, 2020-08-30:
Hi alf45tar you did really great job with this project. I like the direction in which v2.0 is going.  I'm using Pedalino to control Line 6 HX Stomp. In v 2.0  the legacy midi output is not working for me (I've checked and it's on in webUI). Another thing I've realised is NewAction button is not working with WebUI Theme's other than default bootstrap (tried it with Edge, Firefox, Opera ).

jimhiggs, 2020-08-31:
I just upgraded to V2 and it is great. Helicon Switch Six works perfectly and I have expression , momentary 2 switch and a momentary latch. All are showing the expected midi messaging on the lcd when pressed. However when I monitor midi out with pocket midi and a midi to usb adapter cable, all buttons give the same output on the monitor. 90 00 00.
I am pretty sure I used the same adapter when testing earlier versions with no problems. Could this be the same issue mentioned above?

alf45tar, 2020-09-05:
Legacy MIDI works again.

@rafm44 New Action is working for me with Safari, Edge and Chrome. Please check if the page is fully loaded. If too many actions are defined please use the pedal filter.

jimhiggs, 2020-09-05:
Confirmed it works for me. Thanks.

rafm44, 2020-09-06:
Legacy MIDI works for me too.  The problem I had with the WebUI (action btn not working etc) was on my side. It was caused by  improperly uploaded firmware. Thanks alf45tar

closed 2020-09-06 by alf45tar