alf45tar/PedalinoMini #114: WiFi connection timeout

mknerr, 2020-07-20:
Not sure if this is an issue or not, but it's happening consistently enough, I thought I would mention it. It seems every other time I boot my pedalino, the Wifi times out. And, on the opposite times, it connects to my network almost immediately.

alf45tar, 2020-07-20:
Check the log. Are there any errors?

mknerr, 2020-07-20:
Yes, as follows:

[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
SYSTEM_EVENT_STA_DISCONNECTED
SmartConfig started
SmartConfig timeout
WPS started
WPS timeout
AP Pedalino-71AE114C started

alf45tar, 2020-07-20:
I need all the log from the beginning of boot to the end.

alf45tar, 2020-07-21:
Did you try to reset to factory default and/or erase flash?

alf45tar, 2020-08-04:
Is still an issue?

mknerr, 2020-08-06:
I have not been noticing it since I've updated. Thank you.

closed 2020-08-06 by mknerr