alf45tar/PedalinoMini #115: Using momentary switch as toggle switch

frankdeath, 2020-07-23:
Issue #54 discussed how to use a momentary switch as a toggle with the old web interface. Is this still possible with the new web interface?

My ultimate goal is to use the momentary switch of a rotary knob to switch between forward and reverse of the Line 6 Helix Looper (see screenshot).

[image]

frankdeath, 2020-07-23:
It looks like subsequent presses of the momentary switch alternate between the "From" and the "To" values. If this is correct, it would be very helpful to have a unique tag for each value.

alf45tar, 2020-08-11:
@frankdeath added the second tag

closed 2020-08-11 by alf45tar