alf45tar/PedalinoMini #116: Wifi model cannot access interface

zhanwu123, 2020-07-27:
> Executing task in folder PedalinoMini-dev-2.0: C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 245576, Internal Free Heap 215632
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 65536
PlatformIO Built Environment: esp32dev
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/
   \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/
Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\
   \/  /__/
                                                                                  (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 60C40A24
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 260944, Internal Free Heap 230884
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      60C40A24
[NVS][Global][Boot Mode]:        3
[NVS][Global][STA SSID]:         ChinaNet-S6PP2S
[NVS][Global][STA Password]:     07146358743
[NVS][Global][AP SSID]:          Pedalino-60C40A24
[NVS][Global][AP Password]:      60C40A24
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:    60C40A24
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  0
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][Blynk Cloud]:      0
[NVS][Global][Blynk Token]:
[NVS][Global][Ladder]:           Ladder 1 Level 497
[NVS][Global][Ladder]:           Ladder 2 Level 660
[NVS][Global][Ladder]:           Ladder 3 Level 752
[NVS][Global][Ladder]:           Ladder 4 Level 816
[NVS][Global][Ladder]:           Ladder 5 Level 876
[NVS][Global][Ladder]:           Ladder 6 Level 945

Boot WIFI
USB MIDI started
DIN MIDI started
Connecting to
SSID        : ChinaNet-S6PP2S
Password    : 07146358743

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

alf45tar, 2020-07-28:
The error is "Brownout detector was triggered". The reason is linked to power. Not enough power.
Check the cable or change the USB port or use a power adapter.

zhanwu123, 2020-07-30:
This is ESP32 bugs, it took me two pieces of chips appear this problem, the solution is an external battery or in stable power supply

closed 2020-08-11 by alf45tar