alf45tar/PedalinoMini #118: Hope to restart option is added to the web user interface

zhanwu123, 2020-08-02:
Thank you very much alf45tar works，Hope to restart option is added to the web user interface。Can send a link iRig BlueBoard video? Thank you very much！！

alf45tar, 2020-08-03:
Check Options page Reboot button.

alf45tar, 2020-08-03:
If you load the 'blueboard' configuration PedalinoMini can emulate the iRig Blueboard midi mode 1 and 2 as indicated on page 9-11 to the [official manual]( https://download.ikmultimedia.com/plugins/Manuals/iRigBlueBoard/iRig-BlueBoard_User_Manual.pdf).

PedaliniMini must be configured with 4 momentary switches (pedal 1-4) and 2 analog expression pedal (pedal 5-6).

Mode 1 is available on bank 1 and mode 2 is available on bank 2.

Of course PedalinoMidi can emulate the iRig Blueboard hardware not only on BLE interface but in all other supported interface.

That's all folks!

closed 2020-08-11 by alf45tar