alf45tar/PedalinoMini #123: Analog values keep jumping

akosbeke, 2020-08-19:
Hi there,

I'm having a hard time setting up an analog pedal because it keeps sending false messages due to the inconsistent ADC. I know that ESP32 chips are notorious for noisy ADCs.

I do have a 10k pull-up resistor on the analog input and I even experimented with a .1uF capacitor to filter out the signal (as suggested here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#adc-api-adc-calibration) but it does not seem to work.

I can see you are using the ResponsiveAnalogRead library, that should reduce the noise.
But with 10 as an activity threshold the readings fluctuate at around (+-)1-2%.

Things get worse in Wifi mode.

I tried to increase the activity threshold (`Controller.h:1411`) and in Wifi mode the false messages stopped at around 50, but to the detriment of resolution so the values jump in big increments when engaging the expression pedal.

Is there a way of getting rid of the noise with either hardware or software? Am I missing something?

Here's a video that shows the issue:
https://youtu.be/O4lxmtiRTw8

alf45tar, 2020-08-19:
ESP32 is noisy but we can improve in 3 ways:
1. Too many jumpers on your breadboard. I know you need them but they don't help. 
2. Check the setup of other pedals. There are interferences between pins in particular with floating pins without a pull up resistor.
3. Replace ResponsiveAnalog library with a digital filter. This is for me.

akosbeke, 2020-08-20:
I got rid of most of the jumpers and replaced them with small U-shaped wires and put pull-up resistors for all the unused and disabled analog pins.

Now it's a little better but still not quite there yet.

Could you please tell me what you mean by digital filter?

Also it feels like that the problem is not with the readings but with my voltage level.
It jumps between 3.19 to 3.21V in Bluetooth only mode and produces an even worse 3.13 to 3.21 in Wifi mode. This is with a USB cable connected to my computer.

Note that I'm not too experienced with the hardware side of things.

rigr, 2020-08-25:
Akosbeke, you need a stable suplly voltage. I'd suggest NOt to use the USB cable from your computer to power the device, better get an extra power supply for stable 5V to power your board. And use a cpacitor parallel to the board. Hope that helps.

akosbeke, 2020-08-25:
Thanks @rigr,

The external power supply + parallel capacitor combo did the trick.

closed 2020-08-25 by akosbeke