alf45tar/PedalinoMini #124: How to guide?

tominokinawa, 2020-08-25:
Well, I'm trying to figure this all out. I have purchased an ESP32, OLED, I have jumper cables, resistors, push buttons, leds. Now how do I put this together? Do I need a breadboard or can I just solder the jumper cables directly to the ESP32? Has anyone written up a blog about how they put theirs together?

Thank you!

alf45tar, 2020-08-25:
Breadboard is for prototype. Soldering is for the final assembly and it is more complex.
If you don't have an idea how to proceed I suggest to start with a breadboard. There are a lot of tutorial out there.

https://www.circuito.io/blog/breadboards/

tominokinawa, 2020-08-26:
Ok, I just ordered a bread board, thank you. I see some ESP32 modules with a large black antenna, is this needed as well?

mknerr, 2020-08-26:
It depends on what esp32 you have, but probably not. Most have the antenna built in.

tominokinawa, 2020-08-27:
I attempted to make connections with a breadboard from the ESP32 to two buttons and the OLED display to test. I built and uploaded with platform io as explained. So far it doesn't look like the display is receiving power or that there is any kind of bluetooth/wifi signal. In case I'm not connecting things right, does anyone have a picture of the correct way to connect everything with a breadboard?

tominokinawa, 2020-09-01:
Update.

I have the OLED powering up now but the board is not sending any bluetooth or wifi. It is a DOIT ESP32 Devkit V1. I've tried uploading the WIFI and Bluetooth build, and the WIFI only and Bluetooth only builds.

alf45tar, 2020-09-01:
It would be great if you can attach the serial log on boot.

davidin73, 2020-10-09:
> Update.
> 
> I have the OLED powering up now but the board is not sending any bluetooth or wifi. It is a DOIT ESP32 Devkit V1. I've tried uploading the WIFI and Bluetooth build, and the WIFI only and Bluetooth only builds.

Hi! Did you solved? I have the same board....

jisv48, 2020-11-01:
Hello, first of all thanks for sharing the project, it is great work from everyone here.  I just finished loading the software to the esp32 with an oled 1.3".  I followed the instructions by installing on a mac with osx 10.14.6 Mojave the Visual studio and platform.  Followed the instructions to load the software as described here; clone, load, build and load the spiff as indicated. The problem I ran into was a constant failure to load into the ESP32, a problem that I solved by adding a 10mf 25v capacitor to the EN pin and ground as recommended. That solved the problem of loading. Once the load finished, I can see pedalino displaying the WiFi logo, followed for: Release button for Normal Boot, Bluetooth only,  WiFi only, Access Point, AP with BLE,  WiFi reset, Ladder configuration and Hold button for Factory Reset and the loop starts again with a small delay. I opened a serial terminal and I got the following:

I tried to use Smart ESPTouch, holding the EN and touching the BOOT buttons. but I have not being able to get pass that.  Can you provide some suggestions?  Attached is the serial output of the ESP32. Thanks in advance

.
[Pedalino-pictures.pdf](https://github.com/alf45tar/PedalinoMini/files/5470238/Pedalino-pictures.pdf)

alf45tar, 2020-11-01:
Who recommended the capacitor? With the capacitor the EN is permanently pressed.
Disconnect the capacitor and press manually the EN boot during loading the firmware.
With some board the EN button must be pressed manually for starting loading the firmware.

jisv48, 2020-11-01:
Hello, thanks for the quick response regarding my questions for assistance.  I experience the problem I described when running the platform load, build and finally the spiff load.  The problem was intermittent and failed randomly during  the steps with a fatal error that it could not load the code, to the point that I could never get a full load.  Since the problem is well known for some of the ESP32 boards, a search using google produce several solutions, here is one:  

https://www.youtube.com/watch?v=SlAG6DFLnBE <https://www.youtube.com/watch?v=SlAG6DFLnBE>  and https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/ <https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/>   and  https://randomnerdtutorials.com/esp32-troubleshooting-guide/ <https://randomnerdtutorials.com/esp32-troubleshooting-guide/>

After I installed the capacitor,  the load was successfully built and the Pedalino came to life, although there were many warnings during the process.

I removed the capacitor after the load and still have the situation I described, where the board is in a boot loop going from the WiFi picture to the hold button for factory default.

I loaded the EspTouch to my iPhone to try to configure the board as you guys indicated but it does not work.

I also tried the WPS from router with the same results

Any thoughts, I’ll send a video of the configuration and some suggest to change the frequency from 80 to 40 and keep you posted.  I’m using a ESP32 VROOM dev kit 1 from doit.

Thanks again

jisv48, 2020-11-12:
Hello alf45tar, same result no change to the process compiling at 40 or 80, or pressing the boot button every time.  The result is the same all the time, the three steps I used to load Pedalino are:  1. build, 2. load and Platformio run --target uploads.
After the load the ESP32 is in a loop as I explained before.  Are there any issues with the VSC with PlatformIO? I also tried Atom with the platformio extension with the same results. Is there a problem with platformio?

The hardware I have is the ESP32 and the OLED only no additional buttons or anything. 

Questions; when te process to build the app it processes all the environments; esp32dev,esp32doit-devkit-v1,esp32doit-devkit-v1-ble,esp32doit-devkit-v1-wifi, ttgo-t-eight,ttgo-t-eight-ble,ttgo-t-eight-wifi. Do I need to compile all of the them or specify only the esp32doit-devkit-1.  I'm a bit confuse about that, could you explain?

Sorry to keep asking for help, I search the board and it appears that there are some other guys having the same problem. 

Thanks again for the help you provide for the awesomized project.

Cheers

alf45tar, 2020-11-12:
You have to build and upload only your environment.
Do not forget to Upload System Image too and I suggest to reset PedalinoMini to factory defaults once uploaded the firmware.

jisv48, 2020-11-12:
Hello alf45tar,  thanks you for the clarification.  I got the app loaded and working, with a minor problem that perhaps you have experience, I loaded the esp32dev environment and I was able to access the configuration web site as you indicated in the documentation.  The only problem I have is the oled screen, see picture. Do you have any ideas that can help to solve the problem. Meanwhile I'll try other configurations and see if the situation is corrected.  Again thank you for your input and the app is greatly appreciated.  

[image]

akosbeke, 2020-11-12:
Hey @jisv48 

Try this in #141 

https://github.com/alf45tar/PedalinoMini/issues/141#issuecomment-706528127

jisv48, 2020-11-12:
That did it, thank you to all of the help provided. I'll continue to refine the process.  Cheers

closed 2021-01-06 by alf45tar