alf45tar/PedalinoMini #125: Does anybody have this working on heltec_wifi_kit_32  with built in 0.96" OLED?

icooper9000, 2020-08-26:
PdealinoMini.  great lloking device, cant wait to have it working,  I have changed board type to heltec_wifi_kit_32. It has a built in SSD1306 dOLED display. Very useful!

I defined display as SSD1306WIRE in DisplayOLED.h, changed SDA = 4 and SCL =15 as it is connected on the heltec board. Nothing on the display. 

There are signs of life though: BTLE can pair with my phone, Pedalino hotspot is present but I cannot connect to it.
Anybody have it working on Heltec board, if so would you minds sharing how yo got it to work?  

cheers coops9000

alf45tar, 2020-08-26:
Check here #8 even if the halter led is not working with the standard library.

Do you know the password of PedalinoMini hotspot? It is reported on the log.

icooper9000, 2020-08-26:
Thanks, the OLED reset on pin 16 before display init did the trick. I get wifi logo followed by bluetooth logo after reset, and then displays remains blank.  I cannot yet get access to web server.

 I can connect to pedalino hotspot (password as 2nd part of the SSID). When connected I have being trying to connect through 192.168.1.4 but still nothing.  Are there any dependencies that could prevent web server running? Currently I have no I/O connected as I wanted to see if I could get it to basically work before hooking up the switches etc.

alf45tar, 2020-08-26:
Under Upload there is Monitor

icooper9000, 2020-08-26:
Thanks for your help  ah ha! The log shows me that ap address is actually 192.168.4.1 not 192.168.1.4 as it say in getting started notes!   I now have access to the web server.

jaylfc, 2020-11-15:
Did you manage to get it working, I tried a while back but couldn't get the screen to work, please help if you did. Thanks!

icooper9000, 2020-11-15:
I did get the display running but there were too many pinout differences to the doit-ESP32  compatible designs to make it worth continuing. 

I did get pedalino working on a doit compatible ESP32 and it’s great and massively capable,  but more than I needed. I had a specific simpler requirement.  I needed just a configurable midi footswitch with wired and Bluetooth LED midi output to interface to my DAW .  I just wrote my own in the end and it works on the HELTEC, with a cheap rotary encoder with integrated push button for set up. No web interface.

jaylfc, 2020-11-15:
Thanks, thought as much. I have other boards now but I haven't got around to testing but they're not as compact as the heltec so thought I'd ask. Well done on your project sounds awesome! :)

ciapamusche, 2020-12-29:
[image]

Another way to getting work the display in heltec WIFI is set it as  SSH1106WIRE 

Is mandatory change OLED_I2C_SDA to 4 and OLED_I2C_SCL to 15

Also is mandatory the gpio 16 reset procedure before the display_init(); command

I notice as you see in the pic 1 pixel shift but is not annoying and is not visible during playing.

alf45tar, 2020-12-30:
Well done

jaylfc, 2020-12-30:
> [image]
>
> Another way to getting work the display in heltec WIFI is set it as  SSH1106WIRE
>
> Is mandatory change OLED_I2C_SDA to 4 and OLED_I2C_SCL to 15
>
> Also is mandatory the gpio 16 reset [...]

Awesome, could you share your files with me by any chance?

alf45tar, 2020-12-30:
I am trying to include the board in the maintstream

jaylfc, 2020-12-30:
Awesome, can’t wait to try it!

alf45tar, 2020-12-30:
I am releasing it in few minutes.
@jaylfc Can you test for me? I don't have the board.

Thanks
alf45tar

jaylfc, 2020-12-30:
More than happy to!

jaylfc, 2020-12-30:
[image]
[image]

Close but no cigar! Happy to see the screen just light up for a change! Mine is the Lora version but I don't think that makes a difference in this use case.

mknerr, 2020-12-30:
I'll try and dig mine out too. Thanks!

jaylfc, 2020-12-30:
The configs used by @ciapamusche above seem to be different, maybe change them to match?

ciapamusche, 2020-12-30:
Try this platformio.ini file.

[platformio]
default_envs = heltec-wifi-kit-32
;workspace_dir = ~/Downloads/PedalinoMini

[common]
build_flags_esp32 =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
;	-D SSD1306WIRE
	-D SSH1106WIRE
	-D WEBSOCKET
	-D DIAGNOSTIC
;	-D NOSMARTCONFIG
	-D NOWPS
;	-D NOWEBCONFIG
	-D NOBLYNK
build_flags_esp32-ble =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
	-D NOWIFI
	-D SSD1306WIRE
;	-D SSH1106WIRE
build_flags_esp32-wifi =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
	-D NOBLE
	-D SSD1306WIRE
;	-D SSH1106WIRE
	-D WEBSOCKET
	-D DIAGNOSTIC
;	-D NOWEBCONFIG
	-D NOWPS
	-D NOBLYNK
build_flags_ttgo_t_eight =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D TTGO_T_EIGHT
;	-DBOARD_HAS_PSRAM
;	-mfix-esp32-psram-cache-issue
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
	-D WEBSOCKET
	-D DIAGNOSTIC
	-D NOSMARTCONFIG
	-D NOWPS
;	-D BATTERY
;	-D BLUFI
;	-D COMPONENT_EMBED_TXTFILES=data/bootstrap.min.css.gz:data/bootstrap.min.js.gz:data/jquery-3.3.1.slim.min.js.gz:data/popper.min.js.gz:data/logo.png:data/floating-labels.css.gz
;	-D NOWIFI
;	-D NOBLE
;	-D NOWEBCONFIG
	-D NOBLYNK
build_flags_ttgo_t_eight-ble =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D TTGO_T_EIGHT
;	-DBOARD_HAS_PSRAM
;	-mfix-esp32-psram-cache-issue
;	-D DEBUG_ESP_PORT=Serial
;	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
	-D NOWIFI
	-D NOWEBCONFIG
	-D NOBLYNK
build_flags_ttgo_t_eight-wifi =
	-D PLATFORMIO_ENV=$PIOENV
	-D PEDALINO_MINI
	-D TTGO_T_EIGHT
;	-DBOARD_HAS_PSRAM
;	-mfix-esp32-psram-cache-issue
	-D DEBUG_ESP_PORT=Serial
	-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
	-D WEBSOCKET
	-D NOBLE
	-D DIAGNOSTIC
;	-D NOWEBCONFIG
	-D NOBLYNK
	-D NOWPS

[env]
platform 	           = espressif32
framework 	           = arduino
board 		           = esp32dev
board_build.f_flash    = 80000000L
board_build.partitions = src/partitions_ota.csv
lib_deps			   = 	https://github.com/bxparks/AceButton
							https://github.com/bblanchon/ArduinoJson#v6.16.0
							https://github.com/rlogiacco/BatterySense
							https://github.com/blynkkk/blynk-library
							https://github.com/me-no-dev/AsyncTCP
							https://github.com/me-no-dev/ESPAsyncWebServer
							https://github.com/ThingPulse/esp8266-oled-ssd1306
							https://github.com/FastLED/FastLED#3.3.3
							https://github.com/MajicDesigns/MD_REncoder
							https://github.com/CNMAT/OSC
							https://github.com/JoaoLopesF/RemoteDebug
							https://github.com/dxinteractive/ResponsiveAnalogRead
							https://github.com/h2zero/NimBLE-Arduino
							https://github.com/FortySevenEffects/arduino_midi_library
						 	https://github.com/lathoub/Arduino-AppleMIDI-Library
							https://github.com/lathoub/Arduino-BLE-MIDI
						 	https://github.com/lathoub/Arduino-ipMIDI
monitor_speed	       = 115200

[env:esp32dev]
build_flags  = ${common.build_flags_esp32}
board_build.partitions = src/partitions_no_ota.csv
upload_port  = COM3
monitor_port = ${env:esp32dev.upload_port}

[env:esp32doit-devkit-v1]
board 		 = esp32doit-devkit-v1
build_flags  = ${common.build_flags_esp32}
board_build.partitions = src/partitions_no_ota.csv
upload_port  = COM3
;upload_port  = COM4
monitor_port = ${env:esp32doit-devkit-v1.upload_port}

[env:esp32doit-devkit-v1-ble]
board 		 = esp32doit-devkit-v1
build_flags  = ${common.build_flags_esp32-ble}
upload_port  = /dev/tty.SLAB_USBtoUART
;upload_port  = COM4
monitor_port = ${env:esp32doit-devkit-v1-ble.upload_port}

[env:esp32doit-devkit-v1-wifi]
board 		 = esp32doit-devkit-v1
build_flags  = ${common.build_flags_esp32-wifi}
upload_port  = /dev/tty.SLAB_USBtoUART
;upload_port  = COM4
monitor_port = ${env:esp32doit-devkit-v1-wifi.upload_port}

[env:ttgo-t-eight]
build_flags  = ${common.build_flags_ttgo_t_eight}
board_build.partitions = src/partitions_no_ota.csv
;upload_port  = 8e2de6b4.local
;upload_port  = 192.168.2.117
; https://www.silabs.com/documents/public/software/Mac_OSX_VCP_Driver.zip
upload_port  = /dev/tty.SLAB_USBtoUART
;upload_port  = COM5
monitor_port = ${env:ttgo-t-eight.upload_port}

[env:ttgo-t-eight-ble]
build_flags  = ${common.build_flags_ttgo_t_eight-ble}
upload_port  = /dev/tty.SLAB_USBtoUART
;upload_port  = COM5
monitor_port = ${env:ttgo-t-eight-ble.upload_port}

[env:ttgo-t-eight-wifi]
build_flags  = ${common.build_flags_ttgo_t_eight-wifi}
upload_port  = /dev/tty.SLAB_USBtoUART
;upload_port  = COM5
monitor_port = ${env:ttgo-t-eight-wifi.upload_port}

[env:heltec-wifi-kit-32]
platform = ${env.platform}
board = heltec_wifi_kit_32
build_flags = ${common.build_flags_esp32}
upload_port = COM3
monitor_port = ${env:heltec-wifi-kit-32.upload_port}

[env:heltec_wifi-kit-32-ble]
platform = ${env.platform}
board = heltec_wifi_kit_32
build_flags = ${common.build_flags_esp32-ble}
upload_port = COM3
monitor_port = ${env:heltec-wifi-kit-32-ble.upload_port}

[env:heltec-wifi-kit-32-wifi]
platform = ${env.platform}
board = heltec_wifi_kit_32
build_flags = ${common.build_flags_esp32-wifi}
upload_port = COM3
monitor_port = ${env:heltec-wifi-kit-32-wifi.upload_port}

moreover you need to modify the init procedure of OLED display as described by alf45tar

from 1111 line of DisplayOLED.h

void display_init()
{ pinMode(16,OUTPUT);
 digitalWrite(16, LOW);
 delay(50);
 digitalWrite(16, HIGH);
  display.init();
  display.flipScreenVertically();
  display.setContrast(255);

let me know...

jaylfc, 2020-12-30:
I did what you said and somehow managed to break the build process!? I will have a play around with a fresh clone when I get the chance, thanks!

ciapamusche, 2020-12-30:
I tried to compile it before send.

Try to manually modify the platformio file without copy paste

jaylfc, 2020-12-30:
Aaah it was probably due to copying and pasting, just realised different workspace dir. Away from computer now but will retry soon , thanks again!

alf45tar, 2020-12-30:
Can someone test the version I released?
It uses the original Heltec library. It is not anymore requested to modify any file for the init procedure.

According to Heltec the display should be a SSD1306 and not a SH1106.

jaylfc, 2020-12-30:
I tried your version but the screen wasn't working properly, I posted photos above. Strange!

jaylfc, 2020-12-30:
Awesome, looking good now! Will test further tomorrow, thanks for your awesome work and efforts!

[image]

jaylfc, 2020-12-31:
One quick question, how could I change to Bluetooth device name from the serial to something more personal?

ciapamusche, 2020-12-31:
I will try tomorrow

Just for complete the situation @alf45tar :

Legacy midi works with pin 4,15
In this case are occupied by OLED of Heltec.

Is possible to manually remap into 21,22 without problem in order to have the full functionality of the board? 

Very thanks for your support

alf45tar, 2020-12-31:
@jaylfc replace host.c_str() here to change the BLE name

https://github.com/alf45tar/PedalinoMini/blob/b0d4d26ff9785021063053c897bb813263687095/src/BLEMidiOut.h#L24

alf45tar, 2020-12-31:
@ciapamusche I already remapped the pins in case you compile the code for heltec-wifi-kit-32 env.
Check also the code or better the new schematic for pin assignments.

ciapamusche, 2020-12-31:
Ok in these day i will connect everything as schematics. Now I uploaded firmware and I confirm display issue is solved.

closed 2021-03-01 by alf45tar