alf45tar/PedalinoMini #126: Possibility to convert all pins to digital

raidolo, 2020-08-30:
Hi!

I finally built my Pedalino Mini.
I designed a box which is a mix of laser cutting and 3d printing: 

[image]

I'm experiencing an issue with the analog pins, I'm using 3.3Komh pull-up resistors, and my Pedalino is configured with 5x Latch2 switches, so I have 5x buttons connected to the digital pins and 5x buttons connected to the analog pins.
When I press the 4th button connected on the analog pin, something weird happens and I have also the 3rd button doing stuff, fluctuating in a weird way, and I'm not able to turn them off.

Is it possible to configure all pins as digital to get rid of this issue?

Thanks a lot in advance.

Cheers,
Alessio.

alf45tar, 2020-08-31:
Wow...It looks very professional.

Using Latch2 the analog pins are configured as digital.
Can you attach the log?

raidolo, 2020-08-31:
Just a correction, the resistors are 10k omhs not 3.3k. ( I confused the resistors used for the oled.... sorry)

Which log? The one from the serial?

alf45tar, 2020-08-31:
yes the serial log.

raidolo, 2020-08-31:
On the analog side I've found a cold solder and it seems I've fixed it. But I've started to see the same strange behavior also on the digital pins. See the video:

https://youtu.be/gNVr6zfyuAo

First button on the top row is pin 25, second button is pin 26. The first two digital pins. There's no external pull-up resistors connected to them.

It seems that the issue starts when using BLE only, it's like the internal pull-up of my esp32 board doesn't work very well.

Do you think that using an external pull-up resistor also on the digital pins could solve the issue?

Regarding the serial log you'll see no input (as in the video when pushing the green button does nothing) or weird stuff like inputs activated by themselves, not much interesting.

Thanks.

raidolo, 2020-08-31:
I've pulled up all the pins except the pin 12, otherwise the ESP32 don't boot. 

Seems working fine now, I'll let you know, I'll test it more.

raidolo, 2020-09-04:
I had to move the button on pin D12 to the D13, with a 10k omh pull-up resistor. 
Leaving the D12 pin floating leads to unpredictable results. I had a weird behavior like before.
As I said, pin D12 cannot be pulled up, otherwise the ESP32 does not boot. 
SO now I'm using all the "analog" pins except D33 and all the digital pins except D12, all with pull up resistor.

I didn't see strange activations since I've done it. If anything new comes up I'll let you know.

I don't know if it's my dev board that has some issues, but I think this is something worth mentioning in the wiring guide as a possible option.  

Cheers,
Alessio.

alf45tar, 2020-09-04:
Digital pins do not need external pull up resistor because if used the internal pull up resistor is used.

raidolo, 2020-09-04:
What do you mean? Is that because in the code they are defined as INPUT_PULLUP and so you mean I’m using a double resistance in this case?

alf45tar, 2020-09-05:
Yes, confirm. Double resistor is not a problem except for pin 12.

closed 2020-11-16 by alf45tar