alf45tar/PedalinoMini #127: Upload Build Guide

alt8800, 2020-09-09:
First of all, thanks for this wonderful project. I manage to put together the circuit as explained on the schematics but I´m having a lot of problems in the build and upload part. It´s good to know that so far I have only made very simple coding on Arduino IDE. I think my main issue is with the proper set up of the PlatformIO. I just can not pass the PlatformIO: Upload command and it seems that the system won't recognize the ESP32 (ESP-WROOM-32) is connected to the USB port.  Thanks in advance and any help will be very appreciated!

alf45tar, 2020-09-10:
Did you change upload_port inside platfomio.ini file before upload?

bsos, 2020-09-15:
Also want to thank you for your great work!

I'm banging my head trying to build this project and getting the following error:
In file included from src\PedalinoMini.cpp:63:0:
src\BLEMidiOut.h:14:37: fatal error: hardware/BLE-MIDI_ESP32.h: No such file or directory
compilation terminated.

I've cloned the project, and it looks like the library dependencies are there, but when I look in the hardware folder I don't see this header file.  Is there something else I need to do to make sure PIO has all the dependent libraries?

This is just the first of multiple build failures, which I'm guessing is related to me missing some dependencies.  Below is a summary:
Environment               Status    Duration
------------------------  --------  ------------
esp32dev                  FAILED    00:00:06.011
esp32doit-devkit-v1       FAILED    00:00:05.981
esp32doit-devkit-v1-ble   FAILED    00:00:06.165
esp32doit-devkit-v1-wifi  SUCCESS   00:00:11.934
ttgo-t-eight              FAILED    00:00:05.940
ttgo-t-eight-ble          FAILED    00:00:05.627
ttgo-t-eight-wifi         SUCCESS   00:00:11.490

RomanKlein777, 2020-09-15:
Hello, This is great project and I have already some improvement ideas...

But I have exactly same issue. My guess #1-  update to platformio 2.0.1 messed up things. I am currently running one device, but compiled on previous version of platformio.  Is it not just some filename renaming sufficient? like BLE-MIDI_ESP32.h to BLE-MIDI_ESP32.h  ? sorry for stupid question...

Another thing looks suspicious:   when running "pio update" - iI see one incompatible warning: 
Updating platformio/contrib-piohome           3.3.0 @ ~3.3.0                     [Up-to-date]
Updating platformio/contrib-pysite            2.37.191020 @ ~2.37.0              [Incompatible 2.38.191020]
Updating platformio/tool-unity                1.20500.200612 @ ~1.20500.0        [Up-to-date]
Updating platformio/tool-scons                4.40001.0 @ ~4.40001.0             [Up-to-date]

Platform Manager
================
Platform espressif32
--------
Updating platformio/espressif32               2.0.0                              [Up-to-date]
Updating platformio/toolchain-xtensa32        2.50200.80 @ ~2.50200.0            [Up-to-date]
Updating platformio/framework-arduinoespressif32 3.10004.200129 @ ~3.10004.191002   [Up-to-date]
Updating platformio/tool-esptoolpy            1.20600.0 @ ~1.20600.0             [Up-to-date]
Updating platformio/tool-mkspiffs             2.230.0 @ ~2.230.0                 [Up-to-date]

Library Manager
===============
Library Storage: C:\Users\romi\.platformio\lib

alf45tar, 2020-09-15:
Thanks @bsos for the notice 
Arduino-BLE-MIDI#feat/2.0.0 has been recently updated with some incompatible changes
I updated PedaliniMini and now we can build it again.

alf45tar, 2020-09-15:
@RomanKlein777 I am open to any idea.

Build errors are not linked to PlatformIO 5.0.1.

RomanKlein777, 2020-09-15:
sorry I am new to Platformio, until now I have only used Arduino IDE and programming sensors to cloud.PlatformIO looks much more capable but also complicated ... 

I was interested in building piano learning LED strip - I found already those cheap Neopixels which have spacing very very close to my 49 keys MIDI controller. But, I see with pedalino mini we have already all music interfaces, hardware and great firmware here. And suddenly you added also Neopixel support :) My suggestion is : why not just connect after our 6 neopixels for pedals other let's say 49 pixel LEDs, and little bit of programming which shows notes received.  For more LEDs, ofcourse more powerfull power supply should be used. But in terms of connection it is piece of cake and some programming effort will be needed. Configuraion shoud be like: set which interface goes to neopixel keys, which range, come color magic, fading, etc...  For the basic functionality I am going to send you at least 2 beers/coffees ;)

I am in fact going to make the pedalinoMini built-in inside the 49 keys controller and use it with iPad - it will be in fact BLE upgrade. Another dream is to make it Keytar and easily add some more controllers/sensors. Currently I have simple photoresistor hooked as analog pedal ... waving hand makes "pedal move ...

I am also looking for the battery percentage display hints in documentation/sourcecode. Looks like analog pin #1 is the one. But how to activate it?

bsos, 2020-09-15:
Thanks @alf45tar! Everything is building now.  

I'm new to this and want to confirm, to upload this to the esp32, we need to specify the right com port in the ini file and then press the ido button for all 7 of the items to flash to the esp32? Sorry to ask for support.

alt8800, 2020-09-21:
> Did you change upload_port inside platfomio.ini file before upload?

I changed the ports on the platformio.ini file with the same results. It was a hardware issue with certain ESP32 boards, nevertheless, the solution was very simple. I put a 2.2uf capacitor between the ground and the EN pin, and got the upload correct and Pedalino running properly. Thanks again!

closed 2021-01-06 by alf45tar