alf45tar/PedalinoMini #129: Possibility to use SSD1309 2.42" OLED

micahvdm, 2020-09-21:
As the title says, I'm wondering if the above would be possible? Would be really nice to have a bigger OLED for a pedal board situation.

alf45tar, 2020-09-22:
PedalinoMini supports SSD1306 and SH1106 128x64 OLED display only.
If SSD1309 is not backward compatible with SSD1306 it doesn't work.

It could be possible adapting https://github.com/ThingPulse/esp8266-oled-ssd1306 library.

wybbbb, 2021-09-13:
[image]
[image]
After testing, the SSD1309 module compatible with SSD1306 can be used normally. I used translation software to translate it, and the meaning may be a little off!

DWSQUIRES, 2021-09-14:
Cool. Did this require any code changes?

wybbbb, 2021-09-14:
> Cool. Did this require any code changes?

Without modifying any code, you can use it directly after connecting

closed 2021-11-09 by alf45tar