alf45tar/PedalinoMini #130: Pedal three trigger modes

zhanwu123, 2020-09-24:
Hello, alf45tar. I'm prepared to pedal1 set to single press, double press and long press three modes, and has enabled the single press, double press and long press, what should I do?
Thank you very much!

zhanwu123, 2020-09-24:
[image]
[image]

alf45tar, 2020-09-24:
Very good. The setup is ok.
Just press pedal 1 to send the configured MIDI messages.

zhanwu123, 2020-09-24:
Now pedal pressed the from is 127 it after work, is 0, there is no action

alf45tar, 2020-09-25:
@zhanwu123 not clear to me what you mean

zhanwu123, 2020-09-25:
Pedal pressed, the value value is 0, then pedal no MIDI information output, the value value of 127 is the MIDI output

alf45tar, 2020-09-25:
With your configuration when you click (press and release) pedal 1 PedalinoMini send CC 60 value 127.
The second time send CC 60 value 0. The third time CC 60 value 127, the fouth CC 60 value 0 and so on.

zhanwu123, 2020-09-26:
Does or does not work, want to turn off the automatic sorting of pedals, behind every action and release.

zhanwu123, 2020-09-29:
Set before
[image]
Save the future
[image]

zhanwu123, 2020-10-10:
Alf45tar hello, I have already completed the PedalinoMini project, is now double-click and long press cannot be applied, don't know what's the problem? Thank alf45tar.

alf45tar, 2020-10-10:
For Pedal 1 create 3 actions. The first on Press, the second on Double Press and third on Long Press.
Assign different midi messages of course.

zhanwu123, 2020-10-12:
I have already done so, but still can't work perfect, because is worth, 0 or 127 can't return.

alf45tar, 2020-11-30:
@zhanwu123 
Did you solved?

closed 2021-01-03 by alf45tar