alf45tar/PedalinoMini #132: TTGO-T-DISPLAY ?

yaouenvoyage, 2020-09-27:
Is it possible to use the TTGO-T-DISPLAY ?

rigr, 2020-09-28:
It can be done, but as far as I can see, there are not enough ports on that board. And you have to re-define the ports for the display.

alf45tar, 2020-09-28:
The IPS display of TTGO-T-DISPLAY is based on ST7789V.
PedalinoMini supports OLED display based on SSD1306 or SH1106 as of today.

anadarey, 2021-02-12:
Is it possible to use TTGO T-Display (I have one on ST7789V) as headless Padelino midi without display ? Just for the purpose of the few fixed MIDI CC buttons and one rotary control.

alf45tar, 2021-02-13:
Yes, it is.

bobbyzero, 2021-04-15:
Since the T-Display is now supported, this issue can probably be closed.

But the schematic for this model only lists the USB midi pins - to find the 'classic' MIDI pins I had to go to pedalino.h :
```
#define USB_MIDI_IN_PIN       GPIO_NUM_21   // SDA
#define USB_MIDI_OUT_PIN      GPIO_NUM_22   // SCL
#define DIN_MIDI_IN_PIN       GPIO_NUM_15
#define DIN_MIDI_OUT_PIN      GPIO_NUM_2
```

alf45tar, 2021-04-16:
TTGO T-Display in working pretty well but it not 100% finished yet.
There is no enough pins to support USB Midi, DIN Midi and leds all together. You can select just 2 of 3.

closed 2021-04-16 by alf45tar