alf45tar/PedalinoMini #137: boot - button - workaoround

rigr, 2020-10-06:
Hello alf45tar,

I built a pedalo mini using a 30 pin devkit. While transplanting it from the breadboard in it's own house I had to realize that the boot-button is not avalable using any of the pins of the devkit-board, so my workaround was to solder a wire to the button - and it works. I thought that it would be more elegant to gain access to the boot-button functionality using for example two buttons at the same time (Like pressing button 1 and button 2) or - even better:  would it be possible to use D2 instead of the boot-button or using them at the same time?
The boot-button has to be pressed quite often to switch between modes. and soldering on the devkit board is not ideal.

Thank you and best regards
Richard

alf45tar, 2020-10-08:
I don't think the boot button should be pressed so often. It should be used only to reset to factory default or configure the resistor ladder. These operations are not used frequently in real use.

However you can remap it to any available pin changing the following line
https://github.com/alf45tar/PedalinoMini/blob/46f40604f3befd1aff69c31ba92c01bfe330d0c2/src/Pedalino.h#L55

rigr, 2020-10-09:
Thank you alf45tar for your help. I use the boot button also to switch between banks (single and double click) and to switch between the display modes (long press).
I will remap and upload for the next version - thank you!

closed 2020-11-16 by alf45tar

MaxZak2020, 2021-01-03:
> However you can remap it to any available pin changing the following line
> https://github.com/alf45tar/PedalinoMini/blob/46f40604f3befd1aff69c31ba92c01bfe330d0c2/src/Pedalino.h#L55

I've made EN and BOOT buttons on front pannel and remapped Boot on this separate Button (pinned to D13) on GPIO_NUM_13  and it works with Bank change and second screen. But **unexpectedly it dosn't** work with booting mode sequence (EN -> BOOT hold) ...

alf45tar, 2021-01-03:
Change the following line too

https://github.com/alf45tar/PedalinoMini/blob/46f40604f3befd1aff69c31ba92c01bfe330d0c2/src/Pedalino.h#L53