alf45tar/PedalinoMini #139: Firmware update

rigr, 2020-10-09:
I hope you do not mind all my questions. But: How to I build the firmware file to upload? Or where can I find it online?
Thank you!

alf45tar, 2020-10-09:
https://github.com/alf45tar/PedalinoMini#how-to-build-and-upload

rigr, 2020-10-09:
Thanks, I built and uploaded already :)
I was asking for how to build the firmware for OTA firmware update or via HTTP (http://pedalino.local/update).
Thank you!

alf45tar, 2020-10-11:
1. Build your env (for example esp32doit-devkit-v1)
2. Connect to http://[pedalino-hostname].local
3. Select file build/esp32doit-devkit-v1/firmware.bin
4. Press Upload

rigr, 2020-10-11:
Thank you very much alf45tar for clarification! Will build that file with the latest version.

closed 2020-10-11 by rigr

MaxZak2020, 2021-01-06:
Try to Update OTA over HTTP - "Update fail"

Building .pio\build\esp32doit-devkit-v1\firmware.bin
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.5% (used 63980 bytes from 327680 bytes)
Flash: [====      ]  39.2% (used 1489418 bytes from 3801088 bytes)
esptool.py v3.0
========================================= [SUCCESS] Took 898.81 seconds =========================================

Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  SUCCESS   00:14:58.811
========================================== 1 succeeded in 00:14:58.811 ========================================== 

Have any ideas?