alf45tar/PedalinoMini #140: DOIT ESP32 DEVKIT-V1 not recognized

davidin73, 2020-10-09:
Hello @alf45tar .
I'm a newbie and I have two problems,if you can help me.
First of all, I have around 40 compiling errors in PIO , I think is related to the second problem, that is my DOIT ESP32 DEVKIT-V1 isn't recognized from windows.
Can you help me please?
Thanks in advice

alf45tar, 2020-10-09:
The 2 errors are not linked.

Attach the compile log. I will have a look to the errors.

If the DOIT ESP32 is not recognised is because you need to install the USB drivers for CP2102 USB chip.
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

davidin73, 2020-10-09:
Ok, when I come at home I retry to compile it and copy/paste the log here.
Thank you very much for your interest, and for the answer.

davidin73, 2020-10-09:
hi, i'm here
i have partially solved, i have compiles and uploaded,and i can connect to the webserver.
but in the serial console appears three errors:
The first two are
`E (1914) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:72] begin(): Mounting SPIFFS failed! Error: -1`
and when i connecto the error that sequencially appears is:
`[E][vfs_api.cpp:22] open(): File system is not mounted`
effectivelly the web UI appears, but without images.
any suggestion?
Thank you very much!

alf45tar, 2020-10-09:
Run “Upload Filesystem Image” task in PlatformIO IDE or use PlatformIO Core (CLI) and 'platformio run --target uploadfs' command

davidin73, 2020-10-09:
You are my hero!!!!
Everything works fine!!!!!!
Thaaaaaaaaanks!!!!!!

davidin73, 2020-10-09:
Just to explain for the other people that can be have the same problem:
Without the correct driver, I can't know the com port. Installed the driver the board was on the COM3 In my case.
Edited the PlatformIO.ini file to assign to the com3 port,all is compiled and uploaded flawlessly, except the webui ,solved with the @alf45tar suggestion.

closed 2020-10-10 by alf45tar