alf45tar/PedalinoMini #141: Display issues

davidin73, 2020-10-10:
Web io and services working,but display shows this :

[image]

akosbeke, 2020-10-10:
@davidin73 try changing these lines:
```
	-D SSD1306WIRE
;	-D SSH1106WIRE
```

to:

```
;	-D SSD1306WIRE
	-D SSH1106WIRE
```

in platformio.ini

davidin73, 2020-10-10:
@akosbeke thanks,when I back home I try.
Cheers

davidin73, 2020-10-10:
Thank you very much!!!! This was the problem.
By the road,I have two stupid questions (be patient, I'm a noob...😅)
1 i don't understand from the schematic where to connect the six buttons
2 ...and if I want 8 buttons, how can I do?

rigr, 2020-10-10:
David, for better structure I think you should open another issue when changing the topic. 
When clicking on schematic 2 it will open a new window where you can click on RAW and see a bigger version of the sheet.
You can connect up to 6 buttons to any port using a voltage ladder as shown in schematic 2:
[image]
If you want 8 buttons you need to use at least two ports to connect them.

davidin73, 2020-10-10:
Ah,okay
Thank you very much. 
Now I open another topic
Cheers

closed 2020-10-10 by davidin73

jisv48, 2020-11-13:
Thanks for tip regarding the OLED definition.  It is appreciated and now is working correctly.  Cheers