alf45tar/PedalinoMini #144: Probably a srupid question about the switches

davidin73, 2020-10-11:
Hi to all of you!
Just a noob question:
I'm trying to simulate a 2 pedalboard following the schematic, and the second switch does nothing.
I have tried and retried, and the only solution was to do a dual single pushbuttons schema.
Where i'm failing?
My goal is to build a 8 switches pedalboard + 1 external exp pedal,but I'm failing something.
Can someone help by giving my a sketch of a 8 pedals schematic?
Thank you very much!

alf45tar, 2020-10-11:
Connect the fist button to D1 and the second button to A1. Configure pedal 1 as Momentary 2.

davidin73, 2020-10-11:
Thanks for your answer.
But,I must put on all the 'a' pins a 220ohms resistance, or simply I put them serially,the one after the second?😅

alf45tar, 2020-10-11:
The pullup resistor should be 10k.
5k is ok too.

davidin73, 2020-10-11:
Ops,yes, 10k,failed to write.
I put them in parallel,right?
Because I have put them as in the schematics (parallel) of the boss fs6 and nothing is happened...

alf45tar, 2020-10-11:
It is not in parallel. It is a pullup resistor connected dato Vcc 3.3V.

davidin73, 2020-10-11:
Yes,I'm referring to the switches connection,apart the 'ring' resistor part.
I have seen now a video about the definition of a pull up and pull down resistor.
Excuse me, I have really some veeeery basic about electronics And software.
I'm trying to do this beautiful project because I think the most of the commercial projects costs too much for the used technology and have some strong limitations, that your project haven't.
Thanks for your patience.

marcochoin, 2020-10-17:
Hi, beginner here, I still don’t understand the wiring of the foot switches. In the specs, it says you can use up to 36 switches but I have no clue how to do it. Anyway, you responded sw1 on a1 and sw2 on d1 and etc. On the schematics, it seems to say 1 pin on a1, one on d1 and one on grd for one sw. I want to use a stomp foot switch and those I found have 9 pin. Any help on the wiring would be appreciated. I searched a lot but still don’t understand unfortunately.

alf45tar, 2020-10-17:
PedalinoMini has 6 ports for pedals.
If you connect to each port a momentary 1 foot switch (like to Boss FS-5U) you can have 6 switches.
Using 6 "momentary 2" foot switches (like the Boss FS-6) you can have 12 switches.
Using 6 "momentary 3" foot switches (like the Digitech FS3X) you can have 18 switches.
Using 6 "voltage ladder" foot switches (like TC-Helicon Switch 6 ) you can connect 36 switches.

If you don't want to use TRS ports you can replicate the wiring of each port without using TRS connectors.

marcochoin, 2020-10-18:
Thanks for that, I "think" I understand what you are saying. I want to use 3pdt foot switches. How many can I use and what type of wiring do I follow ?

Thanks.

alf45tar, 2020-10-18:
A 3PDT “3 Poles Double Through” is a "momentary 1" switch with 3 separate poles switched together.

There are actually 3 switches in the same box, each of these switches has a midpoint and a flip-flop. This switch will go up or down depending on how many times you press the switch. Thus a contact is made with the pins middle + top OR middle + bottom.

The central pins are either connected to the lower pins or to the upper pins. The footswitch is pressed to toggle the switches.

marcochoin, 2020-10-18:
Thanks for that,

You say it's like a "momentary 1" switch. Can I add multiple switches on
the same port like in the "momentary 3" schema if I wire them correctly?

Thanks.

marcochoin, 2020-10-20:
I have another question.

I don't understand this part

[image]

On other sites, they directly connect pin to pin to the oled display. Why is there a resistor there ? 

Thanks !

alf45tar, 2020-10-20:
The I2C bus pull-up resistor are required by specification. Most of the time they are installed on OLED display board.
You can avoid them but if you note some issue with the display I suggest to try to add them.

Regards
ald45tar

https://www.ti.com/lit/an/slva689/slva689.pdf?ts=1603174931399&ref_url=https%253A%252F%252Fwww.google.com%252F

closed 2020-11-16 by alf45tar