alf45tar/PedalinoMini #145: build problem with latest version

rigr, 2020-10-12:
Hello alf45tar,

when trying to build I get an error:

```
Compiling .pio\build\esp32doit-devkit-v1\src\PedalinoMini.cpp.o
In file included from src\PedalinoMini.cpp:63:0:
src\BLEMidiOut.h:13:21: fatal error: BLEMIDI.h: No such file or directory

*****************************************************************
* Looking for BLEMIDI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:BLEMIDI.h"
* Web  > https://platformio.org/lib/search?query=header:BLEMIDI.h
*
*****************************************************************

compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\libc0c\NimBLE-Arduino\NimBLEService.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libc0c\NimBLE-Arduino\NimBLEUUID.cpp.o
*** [.pio\build\esp32doit-devkit-v1\src\PedalinoMini.cpp.o] Error 1
Compiling .pio\build\esp32doit-devkit-v1\libc0c\NimBLE-Arduino\NimBLEUtils.cpp.o
============================================== [FAILED] Took 14.03 seconds ==============================================
Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  FAILED    00:00:14.031
========================================= 1 failed, 0 succeeded in 00:00:14.031 =========================================The terminal process "C:\Users\Boss\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32doit-devkit-v1'" terminated with exit code: 1.
```

As far as I can see the only thing that changed was 
from 
```
13   #include <BLE-MIDI.h>
14   #include <hardware/ESP32_NimBLE.h>

```
to
```
13   include <BLEMIDI.h>
14   #include <hardware/BLEMIDI_ESP32_NimBLE.h>

```
in  src/BLEMidiOut.h

and 

from 
```
15   #include <BLE-MIDI.h>
16   #include <hardware/ESP32_NimBLE.h>
```

to
```
15   include <BLEMIDI.h>
16   #include <hardware/BLEMIDI_ESP32_NimBLE.h>
```
in  src/BLEMidiIn.h

But changing back to BLE-MIDI.h does not help.

I tried on two different machines where I was able to build the previous version without a problem and now I can not solve this issue alone. Are you able to help?
Thank you and best regards
Richard

alf45tar, 2020-10-12:
Check the latest version just released

rigr, 2020-10-12:
```
================================= [SUCCESS] Took 135.02 seconds ===============================
Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  SUCCESS   00:02:15.022
================================== 1 succeeded in 00:02:15.022 ==================================
```

wonderful, thank you! :))

closed 2020-10-12 by rigr