alf45tar/PedalinoMini #146: The Configuration files can be downloaded and Shared with others PedalinoMini ™ users

zhanwu123, 2020-10-14:
The Configuration files can be downloaded and Shared with others PedalinoMini ™ users. How can I operate?

rigr, 2020-10-14:
well, if you "program" your pedalino (using the webinterface) you can save the configuration to a file and offer that file to others. If you get such a file you can upload it to your pedalino.

zhanwu123, 2020-10-14:
I still can't work. Could you tell me the specific steps? Thank you very much!

rigr, 2020-10-14:
zhanwu123 please explain where you have a problem.

zhanwu123, 2020-10-15:
Configuration file export and import of CFG, what should I do?

rigr, 2020-10-15:
ok, lets start systematically:
1) So you have built a pedalino?
2) Have you been able to call the webinterface?
3) Did you make any settings (actions, pedals, etc?)
next steps later :)

rigr, 2020-10-15:
ok, and did you go to pedals page and actions-page  and change anything?
When done go to configurations- page, type in a name at new configuration (for example zhanwu001), then click "save configuration". 
then choose your zhanwoo001 configuration at "available configurations" and click "download"
you will receive a zhanwu001.cfg file. 
Yes?

zhanwu123, 2020-10-15:
Yes, I can do this.

rigr, 2020-10-15:
ok, so you can then upload that file to your pedalino later. Choose that file at "Upload configuration", upload, then go up to "available configurations" and click "apply and save". That's it.

rigr, 2020-10-15:
and you can send that file to me, and I can also upload to my pedalino. :))

zhanwu123, 2020-10-15:
How to export zhanwu001 this file?

rigr, 2020-10-15:
have you been able to download the zhanwu001.cfg file?

_

> When done go to configurations- page, type in a name at new configuration (for example zhanwu001), then click "save configuration".
> then choose your zhanwoo001 configuration at "available configurations" and click "download"
> you will receive a zhanwu001.cfg file.

_

zhanwu123, 2020-10-15:
Zhanwu001. CFG this file on the phone?

rigr, 2020-10-15:
Did you download the zhanwu001.cfg filr to your phone?
If yes, you can attach it to your reply. I would suggest you use a tablet or even better your computer but it will work with your phone, too.

zhanwu123, 2020-10-15:
[image]
Then what should I do?

rigr, 2020-10-15:
Do you have zhanwu001.cfg on your phone?

rigr, 2020-10-15:
if you want to upload it to here, you will need to rename the file to zhanwu001.cfg.txt or pack it into a zip-file - otherwise github will refuse to accept the file.

zhanwu123, 2020-10-16:
I'm sorry. I still can't import and export CFG file.

zhanwu123, 2020-10-16:
[image]
I've been to the mailbox, but on the computer.

rigr, 2020-10-16:
pk, you downloaded with your browser.

Please have a look here:
1) save configuration inside pedalino
[image]
2) choose configuration
[image]
3) download configuration
[image]
4) decide where to save configuration on your computer
[image]
5) to upload a configuration you need to choose it from your computer
[image]

Please try that. Good luck.

zhanwu123, 2020-10-17:
Why I don't have the item 5？

zhanwu123, 2020-10-17:
After I click on the download directly into the WEB.
[image]

rigr, 2020-10-17:
It might depend on your browser - it looks like your browser is showing what is written in the cnf - file, and you might try right-clicking on "download" (in 3.) and tell your browser where to save the file.
BTW: I suggest you use a better password for your WiFi :)) a123456789 might be too weak.

closed 2020-11-16 by alf45tar