alf45tar/PedalinoMini #148: Error when select bluetooth: CRIT NimBLEAdvertising: "error setting advertisement data; rc=4, "

SteloPin, 2020-11-07:
Hi alf45tar,
Thank you for sharing such a great project!

I can compile and upload the sketch, but only get it running by selecting Wifi only at Boot.
But for my purpose I need the bluetooth function.

Even to compile only for bluetooth by using -ble I get the following error with reboot on serial-monitor (see more lines at the end):
CRIT NimBLEAdvertising: "error setting advertisement data; rc=4, "
abort() was called at PC 0x4010c0ac on core 1

Can you advice what to do?
Thank you and have a greate day.

PS: I'm use this ESP32-model and tried with two boards:
ESP32-DevKitC Core Board ESP32 Development Board ESP32-WROOM-32D ESP32-WROOM-32U Flash 4MB 5V/9V Wireless WiFi Module 
https://www.aliexpress.com/item/ESP32-DevKitC-Core-Board-ESP32-Development-Board-ESP32-WROOM-32D-ESP32-WROOM-32U-Flash-4MB-5V/4001207908183.html?spm=a2g0s.9042311.0.0.2cd54c4dolCXLy

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ChipRevision 1, CPU Freq 240 MHz, SDK Version v3.2.3-14-gd3e562907
Flash Size 4194304, Flash Speed 80000000 Hz
Internal Total Heap 256092, Internal Free Heap 224736
PSRAM Total Heap 0, PSRAM Free Heap 0
nvs Total 65536
PlatformIO Built Environment: esp32dev
_________           .___      .__  .__                 _____  .__       .__     ___ ________________    ___
\______  \ ____   __| _/____  |  | |__| ____   ____   /     \ |__| ____ |__|   /  / \__    ___/     \   \  \
|     ___// __ \ / __ |\__  \ |  | |  |/    \ /  _ \ /  \ /  \|  |/    \|  |  /  /    |    | /  \ /  \   \  \ 
|    |   \  ___// /_/ | / __ \|  |_|  |   |  (  <_> )    Y    \  |   |  \  | (  (     |    |/    Y    \   )  )
|____|    \___  >____ |(____  /____/__|___|  /\____/\____|__  /__|___|  /__|  \  \    |____|\____|__  /  /  /
              \/     \/     \/             \/               \/        \/       \__\                 \/  /__/ 
                                                                                  (c) 2018-2020 alf45star
                                                                      https://github.com/alf45tar/PedalinoMini

Hostname: 5C1C5210
PSRAM not found
Bluetooth Classic disabled: 15252 bytes released
Internal Total Heap 271460, Internal Free Heap 239988
PSRAM Total Heap 0, PSRAM Free Heap 0
SPIFFS mount OK
'nvs' partition was successfully initialized
Reading NVS Global ... done
[NVS][Global][Device Name]:      5C1C5210
[NVS][Global][Boot Mode]:        1
[NVS][Global][STA SSID]:         lo270368
[NVS][Global][STA Password]:     xxxxxxx
[NVS][Global][AP SSID]:          Pedalino-5C1C5210
[NVS][Global][AP Password]:
[NVS][Global][HTTP Username]:    admin
[NVS][Global][HTTP Password]:
[NVS][Global][Bootstrap Theme]:  bootstrap
[NVS][Global][Current Profile]:  2
[NVS][Global][Tap Dance Mode]:   0
[NVS][Global][Bank Switch]:      0
[NVS][Global][Single Time]:      200
[NVS][Global][Double Time]:      400
[NVS][Global][Long   Time]:      500
[NVS][Global][Repeat Time]:      500
[NVS][Global][Encoder Sensit]:   5
[NVS][Global][LedsOnBright]:     5
[NVS][Global][LedsOffBright]:    1
[NVS][Global][Blynk Cloud]:      0
[NVS][Global][Blynk Token]:
[NVS][Global][Ladder]:           Ladder 1 Level 352
[NVS][Global][Ladder]:           Ladder 2 Level 533
[NVS][Global][Ladder]:           Ladder 3 Level 640
[NVS][Global][Ladder]:           Ladder 4 Level 723
[NVS][Global][Ladder]:           Ladder 5 Level 806
[NVS][Global][Ladder]:           Ladder 6 Level 908

Boot NORMAL
USB MIDI started
DIN MIDI started
CRIT NimBLEAdvertising: "error setting advertisement data; rc=4, "
abort() was called at PC 0x4010c0ac on core 1

Backtrace: 0x40092558:0x3ffcfd30 0x40092789:0x3ffcfd50 0x4010c0ac:0x3ffcfd70 0x400d8fa9:0x3ffcfda0 0x400d924a:0x3ffcfe00 0x400f78c9:0x3ffcfe20 0x40125a97:0x3ffcfe80 0x4008e7bd:0x3ffcfea0

Rebooting...
ets Jun  8 2016 00:22:57

alf45tar, 2020-11-08:
It looks like is an issue with NimBLE-Arduino library.
It has been corrected yesterday. Try again now.

https://github.com/h2zero/NimBLE-Arduino/commit/cdb7d7839c1406c71cd8a158d28c11f2a970ded1

SteloPin, 2020-11-08:
Hi alf45tar,
thank you for your quick reply! You are my hero!
I'm a NOOB in PLATFORMIO, so I could not get in sync with the changes on GIT.
Instead I just copied the content of file
https://github.com/h2zero/NimBLE-Arduino/blob/cdb7d7839c1406c71cd8a158d28c11f2a970ded1/src/NimBLEAdvertising.cpp
into my local file NimBLEAdvertising.cpp and compiled again.
After that the rebooting error was gone and I could see the ESP32 as bluetooth device on my iPad.
Thank you again for your help and this fantastic project!
Stefan

closed 2020-11-08 by SteloPin