alf45tar/PedalinoMini #149: How to connect PedalinoMini direct to Fender Mustang or Line 6 POD GO?

SteloPin, 2020-11-09:
Hi,
my Line 6 POD GO does not provide any MIDI-In or Output-Jacks. Instead I can connect it via USB-Cable to my iPad and send MIDI-Commands over that connection.
I would like to connect PedalinoMini via USB to my POD GO and send the MIDI-Commands via bluetooth from the iPad. So I do not need the USB-Cable.

Can this be done via the USB Host Mini-Hardware ? https://www.hackster.io/139994/plug-any-usb-device-on-an-esp8266-e0ca8a
What needs to be done for that, or any other solution?

Thank you, Stefan

alf45tar, 2020-11-09:
It looks like USB Host Mini is designed to mate with Arduino Pro Mini.

https://chome.nerpa.tech/usb-host-shield-hardware-manual/

With the correct library and with some PedalinoMini software modification it can works.

SteloPin, 2020-11-09:
Thank you for your comment alf45star. Would be cool to see it in the future 😄👍

touchgadget, 2021-12-02:
I have not tried PedalinoMini yet but there are other USB Host MIDI to/from UART programs for Adafruit Trinket M0.

https://hackaday.io/project/161715-midi-usb-host-to-midi-din-converter

ESP32 S2 and S3 have USB OTG hardware but USB software support is lagging.

renne, 2021-12-02:
USB-Host support would be great to make my [Fishman Triple Play Connect](https://www.fishman.com/portfolio/tripleplay-connect-midi-guitar-controller/) wireless with additional switches/potentiometers for e.g. octave-shift, pitch, bend/tremolo, etc.