alf45tar/PedalinoMini #152: Problems simulating TC-Helicon Switch 6

flynnguy, 2020-11-17:
I am experimenting with the PedalinoMini and I was able to simulate a single switch but I'm having trouble trying to basically make my own 6 button ladder switch. I used the diagram as an example and I searched for related issues which talked about calibrating the ladder but the interface seems to have changed since then. 

So I have a breadboard setup, even going to Pin's A4 and D4 consistent with the diagrams (though that shouldn't matter as long as I choose the correct pedal if I understand this correctly). I then went into the pedals configuration and set pedal 4 as a ladder and selected single press, leaving function blank and everything else as defaults.

I then went into actions and did new action, pedal 4 and created 6 actions, setting each `on button` `Press` and tried different send commands. I even set the serial port to monitor. I can see certain buttons getting "pressed" though sometimes it also presses a different button on release. I'm guessing this is some sort of calibration issue? I also suspect I'm doing something wrong. Any help would be greatly appreciated.

alf45tar, 2020-11-18:
Calibrate your ladder using booting mode 7 

https://github.com/alf45tar/PedalinoMini#booting-modes

flynnguy, 2020-11-18:
Thanks, don't know how I missed that, works perfectly!

closed 2020-11-18 by flynnguy