alf45tar/PedalinoMini #153: Boot button functionality on a 38pin DevKit v1

copych, 2020-11-23:
Hello, it was so cool to find what I've been thinking of! Great project, nice interface.
By coincidence I've bought a couple of boards to play with, and they should do as I understand. I have ESP32 DEVKIT V1 38pin version, not sure if it's a clone.
Builtin led is gpio1, EN pin is bound to RST button, and gpio0 to BOOT button.
The issue is that the boot button ain't working as expected in this project (
Debug shows that digitalRead(0) always returns 0. No matter if it's pressed or not. I tried different combinations, but not succeeded.
Simple test app i wrote confirmed that the BOOT button is working ok if i test it isolated.
Please, give me a direction to dig.
I've killed a couple of evenings trying to fix it so far, but I only manage to pass the dead loop on startup when remapped the boot btn to external button on another pin, or inversed high/low and pulled gpio0 manually.

copych, 2020-11-25:
I don't know why, but the gpio0 become heavily pulled down right some 100 msecs after reset, so only shorting it to the 3v3 changes the state (. 
But i believe that it is somehow software caused. As i told, simple app can fteely operate with gpio0, lighting the led on gpio1...

RomanKlein777, 2020-11-28:
I have the same problem. my devkit board was running fine with previous version, (by the way it was only 160MHz CPU). now updated (also system image) and only loop. it makes always factory reset as if I was holding BOOT button. I have tried also other boards. Correct 240MHz cpu. No PSRAM versions. Maybe i try those boards. I hope it makes also some difference with PSRAM... 
is it not a bug?

copych, 2020-11-28:
I've remapped both CENTER and BOOT buttons to GPIO23. This way the board behaves more or less predictable. Also I had to update NimBLE library because the included version caused CRI reboots. In fact i have deleted the supplied version and added according section with the github link to the lib_deps section of the project in PlatformIO.
Maybe the author have some thoughts on this matter. For example, could this be some changes that are not properly synced with github?

alf45tar, 2020-11-29:
The issue is with the 3rd parties libraries. I am looking into the issue.

alf45tar, 2020-11-29:
To fix the NimBLE advertising error add the following line to `lib_deps` in platform.ini

https://github.com/h2zero/NimBLE-Arduino

I will release it to `master` branch soon.

alf45tar, 2020-11-29:
Also the BOOT button always LOW depends by a 3rd party library. In this case FastLED. Returned to version FastLED 3.3.3.

closed 2020-11-29 by alf45tar